rbonghi/jetson_stats #7: Jtop not working

mlbach, 2019-05-14:
Thanks for building this useful tool. Unfortunately, `jtop` doesn't run on my Jetson. I get this error:
> Exception in thread Thread-1:
> Traceback (most recent call last):
> File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner    self.run()
> File "/etc/jetson_easy/jtop/jtoplib.py", line 69, in run self.jetsonstats = self.decode(tegrastats_data)
> File "/etc/jetson_easy/jtop/jtoplib.py", line 217, in decode                                                                                   lines = nvpmodel.split("\n")
> TypeError: a bytes-like object is required, not 'str'

Here's my output 'from jetson_release':

> - NVIDIA Jetson TX2
>    * Jetpack 3.3 | 3.2.1 [L4T 28.2.1]
>    * CUDA GPU architecture 6.2
>    * NV Power Mode: MAXN - Type: 0
> - Libraries:
>    * CUDA 9.0.252
>    * cuDNN 7.1.5.14-1+cuda9.0
>    * TensorRT 4.0.2.0-1+cuda9.0
>    * Visionworks NOT_INSTALLED
>    * OpenCV 3.3.1 compiled CUDA: NO
> - Jetson Performance: inactive
> - Jetson Easy v1.5.2

rbonghi, 2019-05-14:
Hi @mlbach ,

thank you to open this issue. I didn't check in these days jtop on a Jetson TX2.
Could you send me the output from:
```
sudo nvpmodel -q
```
Maybe the output is not the same from the Jetson AGX Xavier and Jetson Nano that I'm testing now.

mlbach, 2019-05-14:
Thanks for getting on this so quickly!
For `sudo nvpmodel -q`

> NV Power Mode: MAXN
> 0

For `sudo nvpmodel -q --verbose`

> NVPM VERB: parsing done for /etc/nvpmodel.conf
> NVPM VERB: Current mode: NV Power Mode: MAXN
> 0
> NVPM VERB: PARAM CPU_ONLINE: ARG CORE_1: PATH /sys/devices/system/cpu/cpu1/online: REAL_VAL: 1 CONF_VAL: 1
> NVPM VERB: PARAM CPU_ONLINE: ARG CORE_2: PATH /sys/devices/system/cpu/cpu2/online: REAL_VAL: 1 CONF_VAL: 1
> NVPM VERB: PARAM CPU_ONLINE: ARG CORE_3: PATH /sys/devices/system/cpu/cpu3/online: REAL_VAL: 1 CONF_VAL: 1
> NVPM VERB: PARAM CPU_ONLINE: ARG CORE_4: PATH /sys/devices/system/cpu/cpu4/online: REAL_VAL: 1 CONF_VAL: 1
> NVPM VERB: PARAM CPU_ONLINE: ARG CORE_5: PATH /sys/devices/system/cpu/cpu5/online: REAL_VAL: 1 CONF_VAL: 1
> NVPM VERB: PARAM CPU_A57: ARG MIN_FREQ: PATH /sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq: REAL_VAL: 2035200 CONF_VAL: 0
> NVPM VERB: PARAM CPU_A57: ARG MAX_FREQ: PATH /sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq: REAL_VAL: 2035200 CONF_VAL: 2147483647
> NVPM VERB: PARAM CPU_DENVER: ARG MIN_FREQ: PATH /sys/devices/system/cpu/cpu1/cpufreq/scaling_min_freq: REAL_VAL: 2035200 CONF_VAL: 0
> NVPM VERB: PARAM CPU_DENVER: ARG MAX_FREQ: PATH /sys/devices/system/cpu/cpu1/cpufreq/scaling_max_freq: REAL_VAL: 2035200 CONF_VAL: 2147483647
> NVPM VERB: PARAM GPU: ARG MIN_FREQ: PATH /sys/devices/17000000.gp10b/devfreq/17000000.gp10b/min_freq: REAL_VAL: 1300500000 CONF_VAL: 0
> NVPM VERB: PARAM GPU: ARG MAX_FREQ: PATH /sys/devices/17000000.gp10b/devfreq/17000000.gp10b/max_freq: REAL_VAL: 1300500000 CONF_VAL: 2147483647
> NVPM VERB: PARAM EMC: ARG MAX_FREQ: PATH /sys/kernel/nvpmodel_emc_cap/emc_iso_cap: REAL_VAL: 0 CONF_VAL: 0

rbonghi, 2019-05-19:
Hi @mlbach ,

I have updated the jetson_stats.

I suggest to uninstall your jetson_stats following:
```
cd jetson_stats
sudo ./install_jetson_stats.sh -uninstall
```
after that 
```
sudo -H pip install jetson-stats
```
If your jtop have the same error, launch jtop in this mode:
```
sudo jtop --debug
```
after a while close and copy and paste here the file generate in the same folder you launch called **jtop.log**

rbonghi, 2019-05-21:
Hi @mlbach ,

I found the error in jtop, was for python3 compare my test with python2.7.

The emulation of the error:
https://travis-ci.org/rbonghi/jetson_stats/jobs/535332726#L253

Now the code is fixed and will be release a new version soon. ;-)

commit 43774098 (2019-05-21): Fix #7 for python3
closed 2019-05-21 by rbonghi