rbonghi/jetson_stats #8: ModuleNotFoundError: No module named 'jtopguilib'

monocongo, 2019-05-21:
Thanks for making this tool available.

I installed `jetson_stats` as instructed (`sudo -H pip install -U jetson-stats`), then I ran `jtop`, only to get the following error:

```
$ sudo jtop
Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 6, in <module>
    from jtop.__main__ import main
  File "/usr/local/lib/python3.6/dist-packages/jtop/__main__.py", line 54, in <module>
    from .jtopgui import JTOPGUI, all_info, GPU, Variables
  File "/usr/local/lib/python3.6/dist-packages/jtop/jtopgui.py", line 31, in <module>
    from jtopguilib import *
ModuleNotFoundError: No module named 'jtopguilib'
```

Some additional info:

```
$ sudo nvpmodel -q
NV Power Mode: MAXN
0
```

Please advise, thanks.

monocongo, 2019-05-21:
BTW I am running this on a Jetson Nano, with NVIDIA Tegra X1 (nvgpu)/integrated.

rbonghi, 2019-05-22:
Hi @monocongo ,

Thank you for your open issue. Yesterday night I release a new version of jetson-stats, in particular with python3.

Try to download the new version of jetson-stats:
```
sudo -H pip install -U jetson-stats
```
Keep me posted if the new release (1.6.7) fix this wrong importing

rbonghi, 2019-05-22:
Hi @monocongo ,

I found the error. It was a wrong import with python3. My next release will fix this error.

I will release between today and tomorrow :-)

monocongo, 2019-05-22:
Thanks for the fast follow up, @rbonghi! Looking forward to using the updated version soon...

commit 99f1ac6a (2019-05-22): Fix #8 Wrong library reference for python3
closed 2019-05-22 by rbonghi