rbonghi/jetson_stats #10: my only complain about JTop

blitzvb, 2019-05-22:
hey Raffaello,

My only complain about this tool, is where your name is shown in the UI. I understand that it have to be in, but maybe  at the top right would be better because when I share a screenshot, it's like I am sharing that with your name on it :)

rbonghi, 2019-05-23:
Hi @blitzvb ,

Good idea, I'm thinking a new layout resolution in my next release. Before to release I will post you some different layout option here.

blitzvb, 2019-05-23:
cool I would be happy to help.

rbonghi, 2019-06-03:
Hi @blitzvb ,

I am testing the new header for jtop:
[image]

This is a first option, saving 1 line and move the information on the "INFO" page:
[image]

Keep me posted if you like this solution

blitzvb, 2019-06-03:
obviously I prefer this version... but I also understand the need to have your name. So in addition to the info tab did you think about putting it in the bottom tab at the right corner ? it will be highly visible but not like if it was you that share a top.

while thinking of that I did notice some improvement proposal :

- not sure if it's because you are running it on a dev machine, but all the column are centered aligned (sensor, temp...) while a left align would be more readable and more clean.
- It's not in this screenshot, but on the CPU progress bar, it's always written INTERACTIVE which kind of bother me as it make the progress position more difficult to spot.
- It would be great to have in the GPU tab, an historic of the CPU too ... so this tab can be renamed to graph or historic ?
- why not moving out from the first screen the localhost/ip information to the info tabs. it would uncluttered the most important screen. Unless your user find it very important to have it ALWAYS on screen while I doubt that it will change over time.

so that was my two cents. Hope it helps.

rbonghi, 2019-06-04:
This is a new test, for first page and info page, For your bullet points I will reply soon ;-)
[image]
[image]

blitzvb, 2019-06-04:
Yeah that work for me. 

It’s a branding that I beleave people will treat as such... while still being visible for your own recognition.

rbonghi, 2019-06-04:
Another little fix for the Voltage area. I think now is simple to read.
[image]

blitzvb, 2019-06-04:
Yeah! Awesome.

Another idea : once the UI is all set. You add a screenshot that includes arrow to point to stuff and explain what each things mean.

rbonghi, 2019-06-04:
> * not sure if it's because you are running it on a dev machine, but all the columns are centered aligned (sensor, temp...) while a left align would be more readable and more clean.

I think now all columns are well centered, compare the previous version is more readable.

> * It's not in this screenshot, but on the CPU progress bar, it's always written INTERACTIVE which kind of bother me as it make the progress position more difficult to spot.

I'm not sure remove this information should be great, and in the same time I don't have a good place to move this information in the info screen.

> * It would be great to have in the GPU tab, an historic of the CPU too ... so this tab can be renamed to graph or historic ?

Should be great! At this time I have an history for each CPU available, but I show only the instantaneous status. I can plot in a new table the status of the single CPU without overlap. In a next release I will add it.
Now I want focus to directly control jetson_clock, nvpmodel and the fan from jtop.

> * why not moving out from the first screen the localhost/ip information to the info tabs. it would uncluttered the most important screen. Unless your user find it very important to have it ALWAYS on screen... while I doubt that as it will not change over time.

Yes already done! :-)

closed 2019-06-04 by rbonghi