rbonghi/jetson_stats #9: NameError

blitzvb, 2019-05-22:
Running the pip version, I get that error :

`Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/jtop/__main__.py", line 162, in main
    curses.wrapper(gui, args, tegra)
  File "/usr/lib/python2.7/curses/wrapper.py", line 43, in wrapper
    return func(stdscr, *args, **kwds)
  File "/usr/local/lib/python2.7/dist-packages/jtop/__main__.py", line 108, in gui
    stat = tegra.read
  File "/usr/local/lib/python2.7/dist-packages/jtop/jtop.py", line 142, in read
    while not self._jetsonstats:
  File "/usr/local/lib/python2.7/dist-packages/jtop/__main__.py", line 65, in signal_handler
    tegra.close()
NameError: global name 'tegra' is not defined
`

rbonghi, 2019-05-22:
Can u try to reinstall with:
```
sudo -H pip install --no-cache-dir jetson-stats --upgrade
```
I don't know but sometimes the command "PostInstall" on setup doesn't work well.

Keep me posted if you fix it

rbonghi, 2019-05-22:
About the error with jetson_release:

I found the path error!
https://github.com/rbonghi/jetson_stats/blob/master/scripts/jetson-release#L40

I will fix soon!

Thank you

blitzvb, 2019-05-22:
awesome! thx

rbonghi, 2019-05-22:
About jetson_release 

Meanwhile I fix on my repo you can manually edit in "/usr/local/bin/jetson-release"

and change the line **40** from:
```
source /etc/jetson_easy/jetson_variables
```
to:
```
source /opt/jetson_stats/jetson_variables
```

blitzvb, 2019-05-22:
your fix, did work perfectly! thx again for that and this must have tool.

commit 787fad11 (2019-05-22): Fix error with CTRL-C close #9
closed 2019-05-22 by rbonghi