rbonghi/jetson_stats #11: Run on Jetson Nano got error

gogo000, 2019-05-23:
sudo jtop --debug
Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/jtop/__main__.py", line 152, in main
    curses.wrapper(gui, args, tegra)
  File "/usr/lib/python2.7/curses/wrapper.py", line 43, in wrapper
    return func(stdscr, *args, **kwds)
  File "/usr/local/lib/python2.7/dist-packages/jtop/__main__.py", line 98, in gui
    pages.draw(stat)
  File "/usr/local/lib/python2.7/dist-packages/jtop/jtopgui.py", line 173, in draw
    page(self.stdscr, stat)
  File "/usr/local/lib/python2.7/dist-packages/jtop/jtopgui.py", line 99, in all_info
    line_counter = plot_CPUs(stdscr, line_counter, jetsonstats['CPU'], width)
  File "/usr/local/lib/python2.7/dist-packages/jtop/jtopguilib.py", line 135, in plot_CPUs
    linear_percent_gauge(stdscr, gauge, max_bar, offest + off_idx, start)
  File "/usr/local/lib/python2.7/dist-packages/jtop/jtopguilib.py", line 115, in linear_percent_gauge
    stdscr.addstr(offset, start + 6 + size_bar - len(grey_part), grey_part, curses.A_DIM)
_curses.error: addstr() returned ERR

rbonghi, 2019-05-23:
Hi @gogo000 ,

This is a know bug, I will fix in the next release.
The size of your shell is too much little, use a resolution bigger.

rbonghi, 2019-06-04:
Added Warning screen for size too much small.

[image]

The new release is coming soon

closed 2019-06-04 by rbonghi