rbonghi/jetson_stats #13: Hardware encoding and decoding information

9cvele3, 2019-05-29:
Is it viable to show information about hardware accelerated encoding and decoding sessions (how many of them are active, encoder/decoder usage) and similar?

rbonghi, 2019-05-29:
Hi @9cvele3 ,

I don't know if this information is it available from the internal statistic of the Tegrastats or other data from Linux.

I will checking soon

rbonghi, 2019-06-06:
Hi @9cvele3 ,

Hi read from the devtalk this thread: [tx2-with-jetpack-3-1-gpu-utilization-issue-for-video-encoding-msenc-nvenc](https://devtalk.nvidia.com/default/topic/1024771/jetson-tx2/tx2-with-jetpack-3-1-gpu-utilization-issue-for-video-encoding-msenc-nvenc/ ) I don't know if it avalable this field for the Jetson Nano, I will test this weekend. ;-)

rbonghi, 2019-06-06:
Ok, I should be found the field **MSENC Y** and **NVDEC Y** I will add in the next release.

https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%2520Linux%2520Driver%2520Package%2520Development%2520Guide%2FAppendixTegraStats.html

blitzvb, 2019-06-07:
great idea.

closed 2019-06-16 by rbonghi