rbonghi/jetson_stats #14: crash

blitzvb, 2019-06-07:
hey @rbonghi 

From time to time I have this crash :

nvidia@tegra-ubuntu:~$ sudo jtop 
[sudo] password for nvidia: 
Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/jtop/__main__.py", line 162, in main
    curses.wrapper(gui, args, tegra)
  File "/usr/lib/python2.7/curses/wrapper.py", line 43, in wrapper
    return func(stdscr, *args, **kwds)
  File "/usr/local/lib/python2.7/dist-packages/jtop/__main__.py", line 115, in gui
    key = stdscr.getch()
  File "/usr/local/lib/python2.7/dist-packages/jtop/__main__.py", line 65, in signal_handler
    tegra.close()
NameError: global name 'tegra' is not defined

rbonghi, 2019-06-07:
Hi @blitzvb ,

try to update jetson-stats with:
```
sudo -H pip install --no-cache-dir jetson-stats --upgrade
```
This bug is fixed from version 1.6.8, you should be have an older version installed.
Keep me posted if after this upgrade, you don't have this error.

Best,
Raffaello

PS: To know the version installed use: `pip freeze | grep jetson-stats`

blitzvb, 2019-06-07:
the bug fixe is worst on my system ;)

pip3 freeze | grep jetson-stats
You are using pip version 8.1.1, however version 19.1.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
nvidia@tegra-ubuntu:~$ pip freeze | grep jetson-stats
You are using pip version 8.1.1, however version 19.1.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
jetson-stats==1.6.10

error at each start :

                                   2.0               File "/usr/lib/40thon2.7/subprocess.py", line 711, in __i204M_  0                       370   3797
                            12%] 204M     errread, errwrite)
                             6%] 204M                         File "/usr/lib/python2.7/subp ocess.py", line 10%] 204M_execute_child
                                        50             raise child_exception
                                                                            OSError: [Errno 2] No such file or directory
                                                                               ERROR:jtop.jtop:Exception occurred
                                 Traceback (most recent call last):
                                                                      0le "/usr/local/lib/python2.7/dist-packages/jtop/jtop.py", line 69, in get_nvpmodel
                                                                             nvpmodel_p = sp.Pope7(['nvpmodel', '-q'],9.0dout=sp.PIPE)                 13
                                                        File "/usr/lib/p2thon2.7/subprocess.py", line 711, in __init__                           32   3730
                                          errread, errwrite)
                                                              File "/usr/lib/python2.7/subprocess.py", line 1343, in _execute_child
                                        25             raise child_exception
                                      1.75                                  OSError: [Errno 2] No such file or directory

rbonghi, 2019-06-07:
It's really really strange :open_mouth: Look like is not well installed the new version of jetson-stats.

> File "/usr/local/lib/python2.7/dist-packages/jtop/**main**.py", line 65, in signal_handler
> tegra.close()
> NameError: global name 'tegra' is not defined
About the issue referenced at the head, Do you read this error only when you close jtop with `CTRL-C`, right?

Anyway I will try to emulate on my NVIDIA Jetson tonight.

blitzvb, 2019-06-07:
no it's all the time when I start jTop with the new version that I installed like that :

`sudo -H pip install --no-cache-dir jetson-stats --upgrade`

I can stay in that state (happy to help) or try to remove it and reinstall ?

rbonghi, 2019-06-07:
First for all, thank you for your help, but yes, sure!
Remove and reinstall! :-)

blitzvb, 2019-06-07:
same error.

FYI, it's on Jetson Tx1 , I don't have nvpmodel.

`                                        File "/usr/lib/40thon2.7/subproces .py", line 711, i19%] 612M_                          600      5
                             4%] 612M 7.5 errread, errwrite)
        |                    5%] 612M                         File "/usr/lib/python2.7/subpr||ess.py", line 33%] 612M_execute_child
                                        25             raise child_exception
                                                                            OSError: [Errno 2] No such file or directory
                                                                               ERROR:jtop.jtop:Exception occurred
                                 Traceback (most recent call last):
                                                                     20le "/usr/local/lib/python2.7/dist-packages/jtop/jtop.py", line 69, in get_nvpmodel
                                                                             nvpmodel_p = sp.Popen(['nvpmodel', '-q'], stdout=sp.PIPE)         535    372
                                     39.5               File "/usr/lib/python2.7/subprocess.py", line 711, in __init__                          753   3542
                                          errread, errwrite)
                                                              File "/usr/lib/python2.7/subprocess.py", line 1343, in _execute_child
                                        00             raise child_exception
                                        7                                   OSError: [Errno 2] No such file or directory

`

rbonghi, 2019-06-07:
Ok! Just seen the bug!
I will fix during this weekend!

commit 45493ff2 (2019-06-08): Fixed #14
closed 2019-06-08 by rbonghi