rbonghi/jetson_stats #17: failed to start jtop

oak-tree, 2019-06-12:
Hey, 
The idea looks great but it failed to run on Jetson Xavier jetpack version 4.2

## Extra logs info 
``` 
  Running setup.py develop for jetson-stats
Successfully installed jetson-stats
   * Copy jetson_variables and jetson_performance
   * Copy jetson_env.sh in /etc/profile.d/
   * Copy jetson_variables and other scripts
DONE!
green@greeneye--dev2:~/stress/jetson_stats$ jtop
ERROR:jtop.jtop:Attribute error
Traceback (most recent call last):
  File "/home/green/stress/jetson_stats/jtop/jtop.py", line 222, in run
    self._jetsonstats = self.decode(tegrastats_data)
  File "/home/green/stress/jetson_stats/jtop/jtop.py", line 429, in decode
    ram_status, text = self._RAM(text)
  File "/home/green/stress/jetson_stats/jtop/jtop.py", line 375, in _RAM
    ram_string = re.search(r'RAM (.+?)B', text).group()
AttributeError: 'NoneType' object has no attribute 'group'
Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 11, in <module>
    load_entry_point('jetson-stats', 'console_scripts', 'jtop')()
  File "/home/green/stress/jetson_stats/jtop/__main__.py", line 152, in main
    curses.wrapper(gui, args, jetson)
  File "/usr/lib/python3.6/curses/__init__.py", line 94, in wrapper
    return func(stdscr, *args, **kwds)
  File "/home/green/stress/jetson_stats/jtop/__main__.py", line 96, in gui
    pages.draw(tegra)
  File "/home/green/stress/jetson_stats/jtop/jtopguilib.py", line 46, in wrapped
    return func(self, *args, **kwargs)
  File "/home/green/stress/jetson_stats/jtop/jtopgui.py", line 192, in draw
    page(self.stdscr, stat)
  File "/home/green/stress/jetson_stats/jtop/jtopgui.py", line 116, in all_info
    line_counter = plot_CPUs(stdscr, line_counter, jetson.stats['CPU'], width)
  File "/home/green/stress/jetson_stats/jtop/jtopguilib.py", line 81, in wrapped
    return func(*args, **kwargs)
  File "/home/green/stress/jetson_stats/jtop/jtopguimenu.py", line 53, in plot_CPUs
    linear_percent_gauge(stdscr, gauge, max_bar, offest + off_idx, start)
  File "/home/green/stress/jetson_stats/jtop/jtopguilib.py", line 81, in wrapped
    return func(*args, **kwargs)
  File "/home/green/stress/jetson_stats/jtop/jtopguilib.py", line 160, in linear_percent_gauge
    stdscr.addstr(offset, start, ("{short_name:" + str(name_size) + "}").format(short_name=gauge['name']), curses.color_pair(color_name))
TypeError: integer argument expected, got float
green@greeneye--dev2:~/stress/jetson_stats$ sudo ./install.sh --uninstall

```

```
sudo jtop --debug
Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 11, in <module>
    load_entry_point('jetson-stats', 'console_scripts', 'jtop')()
  File "/home/green/stress/jetson_stats/jtop/__main__.py", line 152, in main
    curses.wrapper(gui, args, jetson)
  File "/usr/lib/python3.6/curses/__init__.py", line 94, in wrapper
    return func(stdscr, *args, **kwds)
  File "/home/green/stress/jetson_stats/jtop/__main__.py", line 96, in gui
    pages.draw(tegra)
  File "/home/green/stress/jetson_stats/jtop/jtopguilib.py", line 46, in wrapped
    return func(self, *args, **kwargs)
  File "/home/green/stress/jetson_stats/jtop/jtopgui.py", line 187, in draw
    self.header()
  File "/home/green/stress/jetson_stats/jtop/jtopguilib.py", line 81, in wrapped
    return func(*args, **kwargs)
  File "/home/green/stress/jetson_stats/jtop/jtopgui.py", line 210, in header
    board_info = os.environ["JETSON_DESCRIPTION"] + " - Jetpack " + os.environ["JETSON_JETPACK"] + " [L4T " + os.environ["JETSON_L4T"] + "]"
  File "/usr/lib/python3.6/os.py", line 669, in __getitem__
    raise KeyError(key) from None
KeyError: 'JETSON_DESCRIPTION'

```

rbonghi, 2019-06-13:
Hi @oak-tree,

how did you install jetson-stats?

I suggest to use:
```elm
sudo -H pip install jetson-stats
```
and do **not** use `./install.sh`

Keep me posted if you fix your error

rbonghi, 2019-06-13:
Already found the bug!

I will fix soon

commit 4c76c28c (2019-06-13): Improve jetson_clocks add test file and Fix #17
commit da27c77b (2019-06-13): Fix #17
closed 2019-06-13 by rbonghi