rbonghi/jetson_stats #20: cannot start  jtop

kenchan13, 2019-06-18:
There is the error I get:
`Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 10, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.6/dist-packages/jtop/__main__.py", line 139, in main
    with jtop(interval=args.refresh) as jetson:
  File "/usr/local/lib/python3.6/dist-packages/jtop/jtop.py", line 81, in __init__
    self.qfan = Fan(path, interval, time)
  File "/usr/local/lib/python3.6/dist-packages/jtop/core/fan.py", line 69, in __init__
    for line in pwm_rpm_table.split("\n"):
TypeError: a bytes-like object is required, not 'str'
`
Thanks!

kenchan13, 2019-06-18:
Output from jetson-release:
```
 - NVIDIA Jetson NANO/TX1
   * Jetpack 4.2 [L4T 32.1.0]
   * CUDA GPU architecture 5.3
 - Libraries:
   * CUDA 10.0.166
   * cuDNN 7.3.1.28-1+cuda10.0
   * TensorRT 5.0.6.3-1+cuda10.0
   * Visionworks 1.6.0.500n
   * OpenCV 3.4.6 compiled CUDA: YES
 - Jetson Performance: inactive

```

output from sudo nvpmodel -q --verbose

```
NVPM VERB: Config file: /etc/nvpmodel.conf
NVPM VERB: parsing done for /etc/nvpmodel.conf
NVPM VERB: Current mode: NV Power Mode: MAXN
0
NVPM VERB: PARAM CPU_ONLINE: ARG CORE_0: PATH /sys/devices/system/cpu/cpu0/online: REAL_VAL: 1 CONF_VAL: 1
NVPM VERB: PARAM CPU_ONLINE: ARG CORE_1: PATH /sys/devices/system/cpu/cpu1/online: REAL_VAL: 1 CONF_VAL: 1
NVPM VERB: PARAM CPU_ONLINE: ARG CORE_2: PATH /sys/devices/system/cpu/cpu2/online: REAL_VAL: 1 CONF_VAL: 1
NVPM VERB: PARAM CPU_ONLINE: ARG CORE_3: PATH /sys/devices/system/cpu/cpu3/online: REAL_VAL: 1 CONF_VAL: 1
NVPM VERB: PARAM CPU_A57: ARG MIN_FREQ: PATH /sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq: REAL_VAL: 102000 CONF_VAL: 0
NVPM VERB: PARAM CPU_A57: ARG MAX_FREQ: PATH /sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq: REAL_VAL: 1428000 CONF_VAL: 2147483647
NVPM VERB: PARAM GPU_POWER_CONTROL_ENABLE: ARG GPU_PWR_CNTL_EN: PATH /sys/devices/gpu.0/power/control: REAL_VAL: auto CONF_VAL: on
NVPM VERB: PARAM GPU: ARG MIN_FREQ: PATH /sys/devices/gpu.0/devfreq/57000000.gpu/min_freq: REAL_VAL: 76800000 CONF_VAL: 0
NVPM VERB: PARAM GPU: ARG MAX_FREQ: PATH /sys/devices/gpu.0/devfreq/57000000.gpu/max_freq: REAL_VAL: 921600000 CONF_VAL: 2147483647
NVPM VERB: PARAM GPU_POWER_CONTROL_DISABLE: ARG GPU_PWR_CNTL_DIS: PATH /sys/devices/gpu.0/power/control: REAL_VAL: auto CONF_VAL: auto
NVPM VERB: PARAM EMC: ARG MAX_FREQ: PATH /sys/kernel/nvpmodel_emc_cap/emc_iso_cap: REAL_VAL: 0 CONF_VAL: 0
```

rbonghi, 2019-06-18:
Hi @kenchan13 

Thanks for your message! I will fix soon!
It is a wrong conversion from in fan controller python (require a decode for python3)

closed 2019-06-18 by rbonghi

rbonghi, 2019-06-18:
Already released the new version, upgrade with (fast way without wait to update the cache in your jetson)
```
sudo -H pip3 install --no-cache-dir -U jetson-stats
```

commit 2afc736e (2019-06-18): Fixed #20
commit 48e12b11 (2019-06-18): New release to fix #20