rbonghi/jetson_stats #21: jtop raises IndexError

wallace11, 2019-06-21:
I saw there were a couple of other complaints around this tool recently.
This is what I get when running on the Jetson Nano:
```sh
$ sudo jtop --debug
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.6/dist-packages/jtop/core/tegrastats.py", line 117, in run
    stats = self.decode(tegrastats_data)
  File "/usr/local/lib/python3.6/dist-packages/jtop/core/tegrastats.py", line 368, in decode
    volt['average'].append(float(value[1]))
IndexError: list index out of range
```

rbonghi, 2019-06-21:
Hi @wallace11 ,

Thank you for your opened issue. I will check and try to emulate your error. Do you have this error every time you open jtop or is occasional?

And really thanks about `dict.get()` I never use it. I totally forget the existence of this method :sweat_smile: .
I'm using a list of if else... I will update my code :+1:

wallace11, 2019-06-21:
Hey,

Thanks for the amazingly fast response. 

The issue happens only when running with `sudo`.

rbonghi, 2019-06-21:
Do you send me the output from `sudo tegrastats`?

wallace11, 2019-06-21:
There you go:
```
RAM 1180/3965MB (lfb 1x2MB) SWAP 149/1024MB (cached 7MB) IRAM 0/252kB(lfb 252kB) CPU [30%@1036,33%@1036,28%@1036,26%@1036] EMC_FREQ 10%@1600 GR3D_FREQ 0%@76 NVENC 716 NVDEC 716 APE 25 PLL@57C CPU@60.5C PMIC@100C GPU@59C AO@65C thermal@59.75C POM_5V_IN 4520/4680 POM_5V_GPU 0/0 POM_5V_CPU 1078/1205
```

rbonghi, 2019-06-22:
I have totally rewritten the Tegrastats parser. Now work fast and I hope without this type of problems!
I will release soon! :-)

commit d045859a (2019-06-22): Refactoring and fix #21
commit e83fa778 (2019-06-22): Add jtop version '-v' and minor fixes #21
closed 2019-06-22 by rbonghi