rbonghi/jetson_stats #29: ZSWAP support.

FPSychotic, 2019-08-10:
Could be possible add stats for  ZSWAP if is activated in kernel? It is a nice peace of software,congrats

rbonghi, 2019-08-10:
Hi @FPSychotic ,

Thank you for your message, and for your issue!
I think it is possible, but I don't know how I can reproduce your scenario.

I'm looking from these pages:
- https://devtalk.nvidia.com/default/topic/1049281/jetson-nano/usb-3-boot-on-jetson-nano/post/5332039/#5332039
- https://syonyk.blogspot.com/2019/04/nvidia-jetson-nano-desktop-use-kernel-builds.html
- https://github.com/Syonyk/raspberry_pi_scripts/blob/master/zswap.sh

This is what are you looking?

FPSychotic, 2019-08-10:
XD, well that is not a issue. I just added that feature to my kernel and used you soft by first time and I thought that maybe interest you.  I saw some commands and a script to get the info, but didn't get them work, as almost everything in Linux XD. Maybe you could be interested in link them here for you?

FPSychotic, 2019-08-10:
Apart, that thing zswap works, is amazing, I could compile Librealsense2 without swap

rbonghi, 2019-08-13:
Hi @FPSychotic ,

sorry if I write late, but I am in holiday :-)
It's a nice idea! I will study a way to implement it, I think is not hard to check, I should be follow this script https://github.com/Syonyk/raspberry_pi_scripts/blob/master/zswap.sh to read this information. 👍

rbonghi, 2020-02-08:
Add extra information to show the zram

[image]

I can show for each zram the used status

FPSychotic, 2020-02-08:
So fast...Thank you!

rbonghi, 2020-02-08:
I added the new feature! :-)
Now you can read the status of each swap directly from the new page

[image]

commit 41e11132 (2020-02-08): Add swap reader. Fix #29
closed 2020-02-09 by rbonghi