rbonghi/jetson_stats #30: Running from terminal support

ghertog, 2019-08-20:
Could it be possible running jtop from a terminal?

From cygwin I get

Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.6/dist-packages/jtop/__main__.py", line 99, in main
    curses.wrapper(JTOPGUI, args.refresh, jetson, [ALL, GPU, CTRL, INFO], init_page=args.page)
  File "/usr/lib/python3.6/curses/__init__.py", line 94, in wrapper
    return func(stdscr, *args, **kwds)
  File "/usr/local/lib/python3.6/dist-packages/jtop/gui/jtopgui.py", line 76, in __init__
    self.run()
  File "/usr/local/lib/python3.6/dist-packages/jtop/gui/jtopgui.py", line 86, in run
    curses.curs_set(0)
_curses.error: curs_set() returned ERR

Removing line 86 from jtopgui.py fixes this. But some characters are strange.

rbonghi, 2019-09-07:
Hi @ghertog ,

I only used with the default terminal in ubuntu 16.04 and 18.04 and works.
I will check soon with cygwin, how I can solve your issue.

ghertog, 2019-09-07:
Ok, thanks in advance

rbonghi, 2020-02-13:
Hi @ghertog ,
I think this issue is fixed from jtop release **1.7.10** and also works with the last one version.

If you have again this error, open this issue

closed 2020-02-13 by rbonghi