rbonghi/jetson_stats #35: Not displaying negative temperatures

gasparka, 2019-10-24:
Reading disappears if below 0.0.

gasparka, 2019-10-24:
We used climate chamber to test Jetson Nano in -20C ambient temperature - nothing bad happened.
Anyways, i was able to read out the negative temperatures manually:
```
cat /sys/devices/virtual/thermal/thermal_zone...
```

Myzhar, 2019-10-24:
Interesting... thank you for your info :+1:

rbonghi, 2019-10-24:
Hi @gasparka ,

I will write you soon. Just to check how I can debug this bug :-)

rbonghi, 2019-10-28:
When you test your board with negative temperatures, can you send me some lines from `sudo tegrastats` ?

gasparka, 2019-10-29:
Here are some lines before negative temperatures:
```bash
RAM 417/3964MB (lfb 727x4MB) SWAP 0/1982MB (cached 0MB) IRAM 0/252kB(lfb 252kB) CPU [0%@1428,0%@1428,0%@1428,0%@1428] EMC_FREQ 0%@1600 GR3D_FREQ 0%@921 APE 25 PLL@15.5C CPU@20C PMIC@100C GPU@17.5C AO@24.5C thermal@18.75C POM_5V_IN 1828/1886 POM_5V_GPU 121/121 POM_5V_CPU 243/297
RAM 417/3964MB (lfb 727x4MB) SWAP 0/1982MB (cached 0MB) IRAM 0/252kB(lfb 252kB) CPU [0%@1428,0%@1428,0%@1428,0%@1428] EMC_FREQ 0%@1600 GR3D_FREQ 0%@921 APE 25 PLL@15C CPU@19.5C PMIC@100C GPU@17.5C AO@24C thermal@18.5C POM_5V_IN 1828/1880 POM_5V_GPU 121/121 POM_5V_CPU 243/291
RAM 417/3964MB (lfb 727x4MB) SWAP 0/1982MB (cached 0MB) IRAM 0/252kB(lfb 252kB) CPU [0%@1428,0%@1428,0%@1428,0%@1428] EMC_FREQ 0%@1600 GR3D_FREQ 0%@921 APE 25 PLL@15C CPU@20C PMIC@100C GPU@17C AO@24.5C thermal@18.5C POM_5V_IN 1828/1876 POM_5V_GPU 121/121 POM_5V_CPU 243/287
RAM 417/3964MB (lfb 727x4MB) SWAP 0/1982MB (cached 0MB) IRAM 0/252kB(lfb 252kB) CPU [0%@1428,0%@1428,0%@1428,0%@1428] EMC_FREQ 0%@1600 GR3D_FREQ 0%@921 APE 25 PLL@15.5C CPU@19.5C PMIC@100C GPU@17.5C AO@24.5C thermal@18.5C POM_5V_IN 1788/1868 POM_5V_GPU 121/121 POM_5V_CPU 243/283
```

And now in negative:
```bash
RAM 808/3964MB (lfb 296x4MB) SWAP 5/1982MB (cached 3MB) IRAM 0/252kB(lfb 252kB) CPU [0%@1428,0%@1428,0%@1428,0%@1428] EMC_FREQ 0%@1600 GR3D_FREQ 0%@921 APE 25 PLL@-9.5C CPU@-5C PMIC@100C GPU@-7.5C AO@-2C thermal@0C POM_5V_IN 1747/1747 POM_5V_GPU 121/121 POM_5V_CPU 162/162
RAM 808/3964MB (lfb 296x4MB) SWAP 5/1982MB (cached 3MB) IRAM 0/252kB(lfb 252kB) CPU [0%@1428,0%@1428,0%@1428,0%@1428] EMC_FREQ 0%@1600 GR3D_FREQ 0%@921 APE 25 PLL@-9.5C CPU@-5C PMIC@100C GPU@-7.5C AO@-1.5C thermal@0C POM_5V_IN 1747/1747 POM_5V_GPU 121/121 POM_5V_CPU 162/162
RAM 808/3964MB (lfb 296x4MB) SWAP 5/1982MB (cached 3MB) IRAM 0/252kB(lfb 252kB) CPU [0%@1428,0%@1428,0%@1428,0%@1428] EMC_FREQ 0%@1600 GR3D_FREQ 0%@921 APE 25 PLL@-9.5C CPU@-5C PMIC@100C GPU@-7.5C AO@-1.5C thermal@0C POM_5V_IN 1747/1747 POM_5V_GPU 121/121 POM_5V_CPU 162/162
RAM 808/3964MB (lfb 296x4MB) SWAP 5/1982MB (cached 3MB) IRAM 0/252kB(lfb 252kB) CPU [1%@1428,0%@1428,0%@1428,0%@1428] EMC_FREQ 0%@1600 GR3D_FREQ 0%@921 APE 25 PLL@-9.5C CPU@-5.5C PMIC@100C GPU@-7.5C AO@-1.5C thermal@0C POM_5V_IN 1747/1747 POM_5V_GPU 121/121 POM_5V_CPU 162/162

```
 Looks like minimum value for "thermal@" is 0. Others go to negative.
Hope this helps.

commit 085bbf4a (2019-11-02): Fix #35
closed 2019-11-02 by rbonghi