rbonghi/jetson_stats #37: Jtop Error Entering CTRL Screen

mdmorar, 2019-11-04:
Hi,

Your last update seems to have broken the jtop ctrl page. Every time I try getting into it it gives me the following error. I am running JetPack 4.2 on the Nvidia Jetson TX2. 

```txt
nvidia@nvidia-desktop:~$ sudo jtop
Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/jtop/__main__.py", line 62, in main
    curses.wrapper(JTOPGUI, args.refresh, jetson, [ALL, GPU, CTRL, INFO], init_page=args.page)
  File "/usr/lib/python2.7/curses/wrapper.py", line 43, in wrapper
    return func(stdscr, *args, **kwds)
  File "/usr/local/lib/python2.7/dist-packages/jtop/gui/jtopgui.py", line 78, in __init__
    self.run()
  File "/usr/local/lib/python2.7/dist-packages/jtop/gui/jtopgui.py", line 99, in run
    self.draw()
  File "/usr/local/lib/python2.7/dist-packages/jtop/gui/jtopguilib.py", line 71, in wrapped
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/jtop/gui/jtopgui.py", line 110, in draw
    page.draw(self.key)
  File "/usr/local/lib/python2.7/dist-packages/jtop/gui/jtopguilib.py", line 107, in wrapped
    return func(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/jtop/gui/pcontrol.py", line 83, in draw
    ctrl_stat = "Enable" if fan['ctrl'] else "Disable" if 'ctrl' in fan else ""
KeyError: 'ctrl'
```

rbonghi, 2019-11-04:
Hi @mdmorar 

I check soon this error,  I keep you posted

closed 2019-11-05 by rbonghi

rbonghi, 2019-11-05:
The new version 1.7.6 should fix the bug. Keep me posted! 

`sudo -H pip install -U jetson-stats`

commit 044b0403 (2019-11-05): Fix #37