rbonghi/jetson_stats #38: does not find tegrastats despite its installed

ardupython, 2019-11-08:
tegrastats is installed under a custom user, which is also the main user running the program.
Tegrastats is located in "/usr/sbin/tegrastats" and can be executed from the commandline as "tegrastats".

But the output of jtop only shows "Tegrastats is not availabe on this board".

Does jtop use a static or dynamic path to tegrastats ?
Where does jtop store the path or how is it created ?
Is there an option to point it to the tegrastats path ?

rbonghi, 2019-11-16:
Hi @ardupython ,

Thank you for your issue, but which board are you using?
jtop automatically find the first tegrastats available in `TEGRASTATS` and is not require to modify the code :-)

> - Does jtop use only the path defined in jtop.py or are there other paths defined else where?

Follow only the paths defined in jtop.py.
Usually tegrastats, jetson_clocks and the fan are available only there when you use an NVIDIA Jetson

> - Is there an option to point it to the tegrastats path besides the TEGRASTATS variable in jtop.py ?

At this time is not available. Do you have another distribution on your jetson?

ardupython, 2019-11-16:
ubuntu 16.04 on a nvidia TX2 Jetson board.
Tegrastats is added to path and can be started directly from the command line, just "tegrastats" with or without sudo.
If i use the default paths, which are defined in  "jtop.py", tegrastats is not found.
As mentioned above, its located in "/usr/sbin/tegrastats", which is not part of the standard paths in "jtop.py".

Are there maybe other environmental variables that could affect it?

rbonghi, 2019-11-16:
With this different configuration I never tried, but without change the code you can try:
* restore jetson-stats code
* Move tegrastats to `/usr/bin/tegrastats` (It's also good to run tegrastats from command line, last new releases of jetpack have tegrastats and jetson_clock in this folder)

> Are there maybe other environmental variables that could affect it?

No, should work fine

I see in your crash you are using this hostname 

> dist-pacuser@vm

Are you working on virtual machine? The name look like that

closed 2020-02-09 by rbonghi