rbonghi/jetson_stats #39: Cannot restore default fan settings

ViorelCN, 2019-11-10:
After using jtop, I cannot restore the default fan settings.

Before I used jtop, the fan came on automatically around 59C and shut off automatically around 43C. After using jtop, this I cannot restore the default settings to go back to this behavior.

Thanks

rbonghi, 2019-11-10:
Hi @ViorelCN ,

Honestly I have the same issue in my board, when you set to "Auto" I set:
1. **/sys/devices/pwm-fan/temp_control** = 1

technically "**temp_control**" enable the temperature control from linux, but I don't know why the fan doesn't wake up over a certain temperature. I finding a way to fix this issue, but it's strange.

Meanwhile you can restore to default following this order
1. `sudo service jetson_performance stop`
2. `sudo bash -c 'echo "1" > /sys/devices/pwm-fan/temp_control' `
3. `sudo bash -c 'echo "0" > /sys/devices/pwm-fan/target_pwm' `
4. `sudo rm /opt/jetson_stats/fan_config` (if you don't have this file is ok)
5. `sudo rm /opt/jetson_stats/l4t_dfs.conf` (if you don't have this file is ok)

ViorelCN, 2019-11-10:
Thanks, will give this a try.

rbonghi, 2019-11-16:
With the new version of jetson-stats, you can use:
```
sudo jtop --restore
```
And will restore all configurations from your jetson.

I temporary disable the option "Auto", I will update in a new version

closed 2019-11-16 by rbonghi