rbonghi/jetson_stats #41: What's the difference among the FAN f "Jc", "Auto" and "Manual"

borninfreedom, 2019-11-15:


rbonghi, 2019-11-15:
Hi @borninfreedom ,

this functionality is under update and fix.

* JC = The fan is controlled from Jetson_clocks
* Auto = the fan speed is automatically controlled via service
* Manual = The fan speed is fixed a manual value

rbonghi, 2019-11-15:
At this time I'm looking to fix this issue #39

borninfreedom, 2019-11-16:
Thank you @rbonghi ,but my jetson nano does not occur the issue #39 even I set the control mode to Auto.  The default auto control mode is not smoothly for my fan, because when the temperature comes to about 59C, the fan rate will reach 31% with a big noisy. So I write the fan a python script, https://blog.csdn.net/bornfree5511/article/details/103076414

borninfreedom, 2019-11-16:
There is another problem I want to consult to you. How does the Jc control mode work?

rbonghi, 2019-11-16:
A next release of jetson-stats will look to include a fan manager, like your!

Usually when you run jetson_clocks the fan is simple manage from the same script:
* temp_control = 0
* fan = 255

If you set "JC" from jtop in this case the fan and fan_control is the same controlled from jetson_clocks

rbonghi, 2019-11-16:
I just release a new version of jetson-stats in this version is temporary disabled the "Auto" mode.
In the new version I will add a service like the article that you post me

borninfreedom, 2019-11-17:
Looking forward to the new version and thank you for your answer.

rbonghi, 2020-02-13:
Same issue on devtalk https://devtalk.nvidia.com/default/topic/1047060/jetson-agx-xavier/fan-management-xavier/post/5424627/#5424627

rbonghi, 2020-02-17:
In according with this topic: https://devtalk.nvidia.com/default/topic/1071493/jetson-agx-xavier/xavier-fan-trouble-re-enable-dvfs-/post/5429430/#5429430

I will retry soon the "Auto" mode and I will enable again this feature on jtop

borninfreedom, 2020-02-26:
OK,thank you!

rbonghi, 2020-06-24:
The new version of jtop library include a simple way to manage the fan.

Finally after a lot of time will be simple to control in 3 ways:

* system (the OS will manage the fan speed)
* jetson_clocks (will follow the usual control by jetson_clocks)
* manual (the fan speed will be controlled manually)

borninfreedom, 2020-06-25:
OK, I will have a try.

rbonghi, 2020-06-28:
New control page.
The 3 options look like more clear than before

[image]

borninfreedom, 2020-06-30:
Cool!

closed 2020-07-13 by rbonghi