rbonghi/jetson_stats #45: jetson_release shows OpenCV 4.1.1 + CUDA installed but only header files are installed

MSBGit, 2020-01-23:
The `jetson_release` script (which depends on output of `jetson_variables` which in turn depends on output of `opencv_version` shows that (on my system) OpenCV 4.1.1 is installed and compiled with CUDA support but I could not find any OpenCV 4.1.1 files other than the header files for C++, no native libraries or Python bindings. This might actually be an `opencv_version` bug but starting here.

```
$ jetson_release
- NVIDIA Jetson NANO/TX1
- Jetpack 4.3 [L4T 32.3.1]
- CUDA GPU architecture 5.3
- Libraries:
- CUDA 10.0.326
- cuDNN 7.6.3.28-1+cuda10.0
- TensorRT 6.0.1.10-1+cuda10.0
- Visionworks 1.6.0.500n
- OpenCV 4.1.1 compiled CUDA: YES
- Jetson Performance: inactive

```

rbonghi, 2020-01-23:
Hi @MSBGit I see now this error.

I'm finding a way to manage that

rbonghi, 2020-02-02:
I should fix your issue, I'm testing. I hope will be ready between today or tomorrow

rbonghi, 2020-02-02:
Fixed from v1.7.10

closed 2020-02-02 by rbonghi