rbonghi/jetson_stats #49: jtop not working

imneonizer, 2020-02-17:
The following error is occurring whenever i run ``sudo jtop``
````
user@localhost:~$ sudo jtop
Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/jtop/__main__.py", line 64, in main
    curses.wrapper(JTOPGUI, args.refresh, jetson, [ALL, GPU, MEM, CTRL, INFO], init_page=args.page)
  File "/usr/lib/python2.7/curses/wrapper.py", line 43, in wrapper
    return func(stdscr, *args, **kwds)
  File "/usr/local/lib/python2.7/dist-packages/jtop/gui/jtopgui.py", line 78, in __init__
    self.run()
  File "/usr/local/lib/python2.7/dist-packages/jtop/gui/jtopgui.py", line 103, in run
    self.draw()
  File "/usr/local/lib/python2.7/dist-packages/jtop/gui/jtopguilib.py", line 71, in wrapped
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/jtop/gui/jtopgui.py", line 110, in draw
    self.header()
  File "/usr/local/lib/python2.7/dist-packages/jtop/gui/jtopguilib.py", line 107, in wrapped
    return func(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/jtop/gui/jtopgui.py", line 139, in header
    set_xterm_title("jtop" + xterm_line(self.jetson))
  File "/usr/local/lib/python2.7/dist-packages/jtop/gui/jtopguilib.py", line 31, in xterm_line
    str_xterm = " " + jetson.board["board"]["Type"]
  File "/usr/local/lib/python2.7/dist-packages/jtop/jtop.py", line 194, in board
    "SN": os.environ["JETSON_SERIAL_NUMBER"].upper()}
  File "/usr/lib/python2.7/UserDict.py", line 40, in __getitem__
    raise KeyError(key)
KeyError: 'JETSON_SERIAL_NUMBER'

````

how ever ``jtop`` without sudo runs fine, it all happened after i upgraded jetson-stats

rbonghi, 2020-02-17:
Hi @imneonizer ,

Which version of jetson-stats are you using?
If you don't know
```
jtop -v
```

Can you try to reboot your board and check it again. ( You helping me to find to bug :-) )

imneonizer, 2020-02-17:
It's ``jtop 1.8.3``

rbonghi, 2020-02-17:
mmm... I'll chek later, yesterday I used it on my Xavier and I didn't see bug.
This is a little little silly bug :-S

Can you open for me the file in **/opt/jetson_stats/jetson_variables** at line 219 you should find something like that:
https://github.com/rbonghi/jetson_stats/blob/master/scripts/jetson_variables#L219

If you don't find that can you share your file here?

imneonizer, 2020-02-17:
I hope this could help you, [jetson_variables.zip](https://github.com/rbonghi/jetson_stats/files/4213754/jetson_variables.zip)

rbonghi, 2020-02-17:
mmm look like when the **pip install** started doesn't update this script.
I will work to understand the reason later.

Meanwhile to don't stop your pipeline, Copy and paste in your jetson_variable all code here: https://github.com/rbonghi/jetson_stats/blob/master/scripts/jetson_variables
Should start to work again with sudo.

imneonizer, 2020-02-17:
Thanks a lot, that helped.
It's working with sudo, now the only thing that is worrying me is:
````
  - Board:                                        Author: Raffaello Bonghi
    * SN:             0423318035020               e-mail: raffaello@rnext.it
    * Jetpack:        4.3 [L4T 32.3.1]
    * Type:           AGX Xavier
    * Name:           NVIDIA Jetson AGX Xavier
    * GPU-Arch:       7.2
  - Libraries:
    * cuDNN:          7.6.3.28-1+cuda10.0
    * VisionWorks:    1.6.0.500n
    * OpenCV:         4.1.1 compiled CUDA: NO
    * CUDA:           10.0.326
    * TensorRT:       6.0.1.10-1+cuda10.0
  - Hostname: smc-test-agx03
  - Interfaces
    * docker0:        172.17.0.1
    * br-b43b6d90eef3:172.21.0.1
    * l4tbr0:         192.168.55.1
    * eth0:           192.168.0.191
````

This ``* OpenCV:         4.1.1 compiled CUDA: NO`` says ``CUDA:NO`` which was ``CUDA: YES`` previously.

rbonghi, 2020-02-17:
> This * OpenCV: 4.1.1 compiled CUDA: NO says CUDA:NO which was CUDA: YES previously.

Should be right not, on version 1.7.8 was a bug: #45 fixed from 1.7.10
You can check if opencv is installed with cuda using:
`opencv_version --verbose`
If is installed with cuda, you will see written there.

imneonizer, 2020-02-17:
I guess it is not compiled with cuda, and on jetson-stats==1.7.8 it was showing the wrong information.
````
General configuration for OpenCV 4.1.1 =====================================
  Version control:               4.1.1-2-gd5a58aa75

  Platform:
    Timestamp:                   2019-12-13T17:25:11Z
    Host:                        Linux 4.9.140-tegra aarch64
    CMake:                       3.10.2
    CMake generator:             Unix Makefiles
    CMake build tool:            /usr/bin/make
    Configuration:               Release

  CPU/HW features:
    Baseline:                    NEON FP16
      required:                  NEON
      disabled:                  VFPV3

  C/C++:
    Built as dynamic libs?:      YES
    C++ Compiler:                /usr/bin/c++  (ver 7.4.0)
    C++ flags (Release):         -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Winit-self -Wsuggest-override -Wno-delete-non-virtual-dtor -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections    -fvisibility=hidden -fvisibility-inlines-hidden -O3 -DNDEBUG  -DNDEBUG
    C++ flags (Debug):           -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Winit-self -Wsuggest-override -Wno-delete-non-virtual-dtor -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections    -fvisibility=hidden -fvisibility-inlines-hidden -g  -O0 -DDEBUG -D_DEBUG
    C Compiler:                  /usr/bin/cc
    C flags (Release):           -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wuninitialized -Winit-self -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections    -fvisibility=hidden -O3 -DNDEBUG  -DNDEBUG
    C flags (Debug):             -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wuninitialized -Winit-self -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections    -fvisibility=hidden -g  -O0 -DDEBUG -D_DEBUG
    Linker flags (Release):      -Wl,--gc-sections  
    Linker flags (Debug):        -Wl,--gc-sections  
    ccache:                      NO
    Precompiled headers:         NO
    Extra dependencies:          dl m pthread rt
    3rdparty dependencies:

  OpenCV modules:
    To be built:                 calib3d core dnn features2d flann gapi highgui imgcodecs imgproc ml objdetect photo python2 python3 stitching ts video videoio
    Disabled:                    world
    Disabled by dependency:      -
    Unavailable:                 java js
    Applications:                tests perf_tests examples apps
    Documentation:               NO
    Non-free algorithms:         NO

  GUI: 
    GTK+:                        YES (ver 2.24.32)
      GThread :                  YES (ver 2.56.4)
      GtkGlExt:                  NO

  Media I/O: 
    ZLib:                        /usr/lib/aarch64-linux-gnu/libz.so (ver 1.2.11)
    JPEG:                        /usr/lib/aarch64-linux-gnu/libjpeg.so (ver 80)
    WEBP:                        build (ver encoder: 0x020e)
    PNG:                         /usr/lib/aarch64-linux-gnu/libpng.so (ver 1.6.34)
    TIFF:                        /usr/lib/aarch64-linux-gnu/libtiff.so (ver 42 / 4.0.9)
    JPEG 2000:                   build (ver 1.900.1)
    HDR:                         YES
    SUNRASTER:                   YES
    PXM:                         YES
    PFM:                         YES

  Video I/O:
    FFMPEG:                      YES
      avcodec:                   YES (57.107.100)
      avformat:                  YES (57.83.100)
      avutil:                    YES (55.78.100)
      swscale:                   YES (4.8.100)
      avresample:                NO
    GStreamer:                   YES (1.14.5)
    v4l/v4l2:                    YES (linux/videodev2.h)

  Parallel framework:            TBB (ver 2017.0 interface 9107)
[... 8 lines cut ...]

  Python 2:
    Interpreter:                 /usr/bin/python2.7 (ver 2.7.15)
    Libraries:                   /usr/lib/aarch64-linux-gnu/libpython2.7.so (ver 2.7.15+)
    numpy:                       /usr/lib/python2.7/dist-packages/numpy/core/include (ver 1.13.3)
    install path:                lib/python2.7/dist-packages/cv2/python-2.7

  Python 3:
    Interpreter:                 /usr/bin/python3 (ver 3.6.9)
    Libraries:                   /usr/lib/aarch64-linux-gnu/libpython3.6m.so (ver 3.6.9)
    numpy:                       /usr/lib/python3/dist-packages/numpy/core/include (ver 1.13.3)
    install path:                lib/python3.6/dist-packages/cv2/python-3.6

  Python (for build):            /usr/bin/python2.7

  Java:                          
    ant:                         NO
    JNI:                         NO
    Java wrappers:               NO
    Java tests:                  NO

  Install to:                    /usr
-----------------------------------------------------------------

````

rbonghi, 2020-02-17:
Yes exactly, now you read the right value on yout jtop!

rbonghi, 2020-02-17:
Use this workaround, works better:
```
sudo -H pip install --no-cache-dir -U jetson_stats
```

rbonghi, 2020-03-28:
From the coming version **2.0.0** will be available `jetson_config` an automatic tool to manage the package status and fix when something doesn't work from **health** page.
This is a preview:
[image]
[image]

closed 2020-03-28 by rbonghi