rbonghi/jetson_stats #51: jtop should not print Total value as sum of POM_5V_IN, POM_5V_GPU and POM_5V_CPU

Lissanro, 2020-03-21:
When running jtop I noticed that Total Power is greater than POM_5V_IN:
[image]
According to https://forums.developer.nvidia.com/t/power-consumption-monitoring/73608/8 "POM_5V_IN measures the overall 5V input to the board, it represents the total power consumption. You don’t need to add POM_5V_GPU and POM_5V_CPU to it. The GPU/CPU rails are provided for extra reference."

That means Total row is always wrong and correct total value is given by POM_5V_IN. I suggest to either remove Total row or make it equal to POM_5V_IN, or rename "POM_5V_IN" to "POM_5V_IN (Total)".

rbonghi, 2020-03-22:
Hi @Lissanro 

Thank you for this hint. At this time the total is evaluated only with a direct sum from all data from tegrastats.

I will find a way to add this correction in my new version of jetson_stats

thank you

rbonghi, 2020-03-25:
I was looking in deep the total power consumption and your issue works for:
* Jetson Nano
* Jetson TX2
* Jetson TX1 (to check)

In according with [Xavier total power consumption](https://forums.developer.nvidia.com/t/xavier-jetson-total-power-consumption/81016) the total power in this case is a sum of all field.

To manage this issue I modify the output with this shape [image below], where in this case the Total power is written in bold.
[image]

This bug will be fix from the coming release **2.0.0**

commit b206023f (2020-03-25): Fix #51
closed 2020-03-28 by rbonghi