rbonghi/jetson_stats #52: does it support execution jtop without elevation to sudo?

AndreV84, 2020-04-04:
does it support execution jtop / jetson-stats without elevation to sudo?

rbonghi, 2020-04-04:
Hi @AndreV84 ,

jtop works without sudo, but not all features are available.
jtop read some information from tegrastats and jetson_clocks to build all menu.

tegrastats and jetson_clocks as well, require to read all information the sudo elevation.

Without use jtop, the only output available is like this picture
[image]

At this time I didn't find any other solution to read the same data without sudo.

AndreV84, 2020-04-04:
Thank you for your response!

rbonghi, 2020-04-04:
You are welcome! :+1:

rbonghi, 2020-06-24:
I’m completely rewriting jtop to works without sudo. I think in 1 or 2 weekS I will release the first release candidate to test.

I will write here a new message when you can start to try.

PR #67 (2020-07-09): Feature/no more sudo
This new feature make jtop available to work without sudo permission.

Now jtop is split in service/client (an example how it is work is in a picture below)

This feature fix #52 and #41 and rearrange the jetson-stats structure to work in docker #63 

[image]
closed 2020-07-13 by rbonghi