rbonghi/jetson_stats #53: ValueError with sudo jtop

AdrienChassignet, 2020-04-07:
Hello @rbonghi 

I am getting started with the Jetson TX2 and I have the following error when running `sudo jtop`:

```
Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.6/dist-packages/jtop/__main__.py", line 92, in main
    curses.wrapper(JTOPGUI, args.refresh, jetson, [ALL, GPU, MEM, CTRL, INFO], init_page=args.page)
  File "/usr/lib/python3.6/curses/__init__.py", line 94, in wrapper
    return func(stdscr, *args, **kwds)
  File "/usr/local/lib/python3.6/dist-packages/jtop/gui/jtopgui.py", line 97, in __init__
    self.run()
  File "/usr/local/lib/python3.6/dist-packages/jtop/gui/jtopgui.py", line 127, in run
    self.draw()
  File "/usr/local/lib/python3.6/dist-packages/jtop/gui/jtopguilib.py", line 71, in wrapped
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/jtop/gui/jtopgui.py", line 138, in draw
    page.draw(self.key)
  File "/usr/local/lib/python3.6/dist-packages/jtop/gui/pall.py", line 161, in draw
    mini(self.stdscr, idx * column_width, line_counter, column_width, self.jetson)
  File "/usr/local/lib/python3.6/dist-packages/jtop/gui/jtopguilib.py", line 107, in wrapped
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/jtop/gui/jtopguimenu.py", line 158, in compact_info
    jc_status = jc.status
  File "/usr/local/lib/python3.6/dist-packages/jtop/core/jetson_clocks.py", line 128, in status
    show = self.show()
  File "/usr/local/lib/python3.6/dist-packages/jtop/core/jetson_clocks.py", line 85, in show
    str(match.group(9)): int(match.group(10))}}
ValueError: invalid literal for int() with base 10: '1 c7=1 '
```

Just running `jtop` gives no error though.
I have the version 2.0.1 and already tried to reinstall with:

```
sudo -H pip3 uninstall jetson-stats
sudo -H pip3 install --no-cache-dir jetson-stats --upgrade
```
This did not solved the problem.
As I saw in similar issues, this might be related to the environment variables so here are mine:

```
~$ env | grep JETSON
JETSON_TYPE=TX2
JETSON_BOARDIDS=3310:0000:A0
JETSON_CUDA_ARCH_BIN=6.2
JETSON_CHIP_ID=24
JETSON_OPENCV=4.1.1
JETSON_L4T_RELEASE=32
JETSON_L4T=32.3.1
JETSON_VISIONWORKS=1.6.0.500n
JETSON_OPENCV_CUDA=NO
JETSON_SOC=tegra186
JETSON_MACHINE=NVIDIA Jetson TX2
JETSON_JETPACK=4.3
JETSON_CODENAME=quill
JETSON_CUDA=10.0.326
JETSON_L4T_REVISION=3.1
JETSON_SERIAL_NUMBER=1420120029923
JETSON_BOARD=P3310-1000
JETSON_TENSORRT=6.0.1.10
JETSON_CUDNN=7.6.3.28
```

Thanks in advance!

rbonghi, 2020-04-07:
Hi thank you for your issue!

Can you send me the output from:
```bash
sudo jetson_clocks --show
```

I will fix soon!

AdrienChassignet, 2020-04-07:
Thank you for your prompt answer!
`sudo jetson_clocks --show` gives :

```
SOC family:tegra186  Machine:quill
Online CPUs: 0,3-5
CPU Cluster Switching: Disabled
cpu0: Online=1 Governor=schedutil MinFreq=345600 MaxFreq=2035200 CurrentFreq=2035200 IdleStates: C1=1 c7=1 
cpu1: Online=0 Governor=schedutil MinFreq=345600 MaxFreq=2035200 CurrentFreq=1113600 IdleStates: C1=1 c6=1 c7=1 
cpu2: Online=0 Governor=schedutil MinFreq=345600 MaxFreq=2035200 CurrentFreq=1113600 IdleStates: C1=1 c6=1 c7=1 
cpu3: Online=1 Governor=schedutil MinFreq=345600 MaxFreq=2035200 CurrentFreq=1881600 IdleStates: C1=1 c7=1 
cpu4: Online=1 Governor=schedutil MinFreq=345600 MaxFreq=2035200 CurrentFreq=2035200 IdleStates: C1=1 c7=1 
cpu5: Online=1 Governor=schedutil MinFreq=345600 MaxFreq=2035200 CurrentFreq=2035200 IdleStates: C1=1 c7=1 
GPU MinFreq=114750000 MaxFreq=1134750000 CurrentFreq=114750000
EMC MinFreq=40800000 MaxFreq=1600000000 CurrentFreq=1600000000 FreqOverride=0
Fan: speed=0
NV Power Mode: MAXP_CORE_ARM
```

rbonghi, 2020-04-07:
Fixed from jetson-stats **2.0.2** :+1:

closed 2020-04-07 by rbonghi

AdrienChassignet, 2020-04-07:
Thank you very much!

Qasim7520, 2023-12-08:
### **#Module Error with sudo jtop**
 sudo jtop
Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.8/dist-packages/jtop/__main__.py", line 159, in main
    curses.wrapper(JTOPGUI, jetson, pages, init_page=args.page,
  File "/usr/lib/python3.8/curses/__init__.py", line 105, in wrapper
    return func(stdscr, *args, **kwds)
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/jtopgui.py", line 100, in __init__
    self.run(loop, seconds)
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/jtopgui.py", line 129, in run
    self.draw()
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/jtopgui.py", line 143, in draw
    page.draw(self.key, self.mouse)
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/pall.py", line 190, in draw
    func(self.stdscr, pos_y_mini_menu, column_width * nline, column_width, column_height, self.jetson)
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/pengine.py", line 106, in compact_engines
    map_eng = map_engines(jetson)
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/pengine.py", line 91, in map_engines
    func_list_engines = engine_model(jetson.board['hardware']["Module"])
KeyError: 'Module

commit fd393167 (2020-04-07): Fix #53