rbonghi/jetson_stats #54: Jetpack missing [L4T 32.4.2]

Robotmad, 2020-04-25:
Please update jetson-stats with new jetpack 4.4

**Linux for Tegra**
 - L4T: 32.4.2

**Jetson-Stats**
 - Version: 2.0.3

rbonghi, 2020-04-25:
Thank you! I will update soon! :-)

rbonghi, 2020-04-28:
The name of this version is Jetpack 4.4 DP (Developer Preview)

Reference [JetPack 4.4 Developer Preview - L4T R32.4.2 released](https://forums.developer.nvidia.com/t/jetpack-4-4-developer-preview-l4t-r32-4-2-released/120594/17?u=rbonghi)

closed 2020-04-28 by rbonghi

rbonghi, 2020-04-28:
New version **2.0.4** released

To update
```bash
sudo -H pip install --no-cache-dir -U jetson_stats
```
To check if the installation is fine and jetson-stats works you can use the health tool running:
```bash
jetson_config
```
Page ***health**

commit 07d31a86 (2020-04-28): Fix #54 with name Jetpack 4.4 DP (Developer Preview)