rbonghi/jetson_stats #55: Jetpack missing [L4T 32.2.3]

naxvm, 2020-04-26:
Please update jetson-stats with new jetpack

**Linux for Tegra**
 - L4T: 32.2.3

**Jetson-Stats**
 - Version: 2.0.3

rbonghi, 2020-04-26:
Hi @naxvm 

Will be fix soon, check this issue #54

rbonghi, 2020-04-26:
I'm sorry! (Sunday afternoon :-D)
 
I was reading now the Linux kernel and I thought was the same of #54.
Can you please me share which Jetpack are you installed?

Should be around Jetpack 4.2.2 and 4.3

naxvm, 2020-04-26:
Hello! Sure, it is Jetpack 4.2.2, jtop has been working properly until today, when I tried to update it.

I would like to thank you as well for the amazing job of these tools! They are really useful for me to control the board.

rbonghi, 2020-04-26:
Thank you! :100: 

Found! You are working on [Jetpack 4.2.3](https://developer.nvidia.com/jetpack-423-archive) :beer: 

Detailed specification are here: https://developer.nvidia.com/embedded/linux-tegra-r3223

I will update soon!
I keep you posted when I release the new jetson-stats update

naxvm, 2020-04-26:
Huge thanks man!! I will keep an eye on it! :)

closed 2020-04-28 by rbonghi

rbonghi, 2020-04-28:
New version **2.0.4** released

To update
```bash
sudo -H pip install --no-cache-dir -U jetson_stats
```
To check if the installation is fine and jetson-stats works you can use the health tool running:
```bash
jetson_config
```
Page ***health**