rbonghi/jetson_stats #56: How to use in virtual env?

js1972, 2020-04-28:
I have installed jetson_stats as per the README with `sudo -H pip install -U jetson-stats`.

I want to use the jetson stats as a library inside my own python script which I develop in a python virtual env. I'm unable to do `from jtop import jtop` as per your examples.

I tried doing the pip install from within my virtual env but it made no difference and I suspect its because of running it with sudo.

Any ideas how I could use jtop in a python virtual environment?

rbonghi, 2020-04-29:
Hi @js1972 

I'm sorry to reply late.
The reason that jetson-stats need to be installed via **sudo** is to have the grant access on tegrastats and jetson_clocks. At this time I didn't found a solution to install without sudo.

It is a really cool enhancement and I hope to close this issue.

At this time I suggest to use this option when you make your virtualenv
```bash
virtualenv --system-site-packages .venv
```
With the option `--system-site-packages` Give the virtual environment access to the global site-packages.

js1972, 2020-05-05:
Thanks for the idea with --system-site-packages. I never knew that one.  ;-)

rbonghi, 2020-05-05:
You are welcome!

I am investigating a solution to install this package without sudo, to make more portable.
Right now I don't want close this issue

DarylWM, 2020-05-10:
I'd like to do the same thing with Archiconda.

rbonghi, 2020-05-10:
Hi @DarylWM ,

I never used Archiconda, I don't know if there is a option to load the packages from the system.

I'm looking a way to install jetson-stats without sudo (I can fix this issue #52 as well), I've to manage a way to install:
- jetson_config
- jetson_release
- jetson_swap

all services:
- jetson_stats_boot
- jetson_performance
And have access in sudo to tegrastats and jetson_clocks

I think there is a way, but I didn't investigtate in deep. I hope for the next release (**not** the coming _2.1_) I will manage it

rbonghi, 2020-06-24:
I’m making a significant update on jetson-stats.

I think for the first stable release of jetson-stats you will be able to install in a virtual environment.

I think probably the first install will need anyway to install via **sudo** all other install sudo will be not required anymore

rbonghi, 2020-07-18:
From the coming version 3.0 jetson-stats will be more simple to integrate on virtualenv.

I cannot close now this issue, because will need another small update. I will keep you posted when will be ready

DarylWM, 2020-07-19:
Thanks for the update Raffaello.

rbonghi, 2021-03-01:
Hi all,

I simply from jetson-stats *3.1.0* the installation in a virtualenv, follow the readme: 
https://github.com/rbonghi/jetson_stats#virtual-environment

You need to install the first time jetson-stats in your host, after that you can reinstall a second time in your virtual environment.

I hope can be simple now for you to use it.

closed 2021-03-01 by rbonghi