rbonghi/jetson_stats #61: Wrong version of ncurses?

DarylWM, 2020-05-23:
Thanks for writing this tool @rbonghi . A question: my display doesn't look as nice as your screenshots, where the layout isn't drawn properly. I'm wondering whether I'm missing the right version of ncurses? Or a terminal setting perhaps?

[image]

rbonghi, 2020-05-23:
Hi @DarylWM 

Wow! yes, I can see that. In which OS are you working?
Look like MacOS (Meanwhile I check on mine)

DarylWM, 2020-05-23:
Hi @rbonghi - yes I'm using macOS 10.15.4.

rbonghi, 2020-05-23:
mmm weird, I don't have problem on mine. I think is something around the lang typeset.
Can you try this test for me?
Try this:
```console
export NCURSES_NO_UTF8_ACS=1
sudo jtop
```

I hope the output should be the same
[image]

DarylWM, 2020-05-23:
I don't think it made a difference.

[image]

rbonghi, 2020-05-23:
Do you have the same configuration?

In **Terminal** > **Profiles** > **Advanced**
[image]

DarylWM, 2020-05-23:
Hi @rbonghi - yes those settings match. I also tried setting LC_ALL=C on the Jetson but that didn't help.

This is what locale returns on the Jetson:
```
LANG=en_AU.UTF-8
LANGUAGE=en_AU:en
LC_CTYPE="en_AU.UTF-8"
LC_NUMERIC="en_AU.UTF-8"
LC_TIME="en_AU.UTF-8"
LC_COLLATE="en_AU.UTF-8"
LC_MONETARY="en_AU.UTF-8"
LC_MESSAGES="en_AU.UTF-8"
LC_PAPER="en_AU.UTF-8"
LC_NAME="en_AU.UTF-8"
LC_ADDRESS="en_AU.UTF-8"
LC_TELEPHONE="en_AU.UTF-8"
LC_MEASUREMENT="en_AU.UTF-8"
LC_IDENTIFICATION="en_AU.UTF-8"
LC_ALL=

```

rbonghi, 2020-05-25:
Thx @DarylWM 

I think is something around the gui host. I have seen time ago on PUTTY a similar problem and I found on stack overflow this solution `export NCURSES_NO_UTF8_ACS=1`.

In our case is tricky to make a setup on my PC to catch this error :disappointed:
We have something different in our PCs, that I don't find it

DarylWM, 2020-05-27:
Thanks @rbonghi . I'll keep an eye out for a fix; it's still a very useful application!

rbonghi, 2020-06-27:
Hi @DarylWM 

I don't know why the reason, but now I reproduce your same error on my laptop.
[image]

I keep you posted when I will fix it

closed 2022-06-11 by rbonghi