rbonghi/jetson_stats #65: I cannot control jtop on page 4MEM to alter the swap file on Jetson NX

Philliec459, 2020-06-03:
**Describe the bug**
I have been running jtop for about 2 weeks now on my Jetson NX on Jetpack 4.3. It works great, but on the 4MEM page I now have no control of the swap space. I could swear in the beginning I could change the swap file size and enable/disable, but now I cannot get out of Extra/Swap and Status/Enabled. I now have a 8GB swap file and 4GB of zram working with 6 CPU's. I have tried to cut back on the swap file to 4GB, but I can't seem to make any changes. 

I have tried sudo jtop --restore , but that did not help. 

I would appreciated any advice on how to handle this.

rbonghi, 2020-06-03:
Hi @Philliec459 ,

I apologise for my late reply.
Yes, you follow a proper way to disable the swap, you can use as well another tool already included in jetson-stats.
```bash
jetson_swap -d [directory] -n [name] --off
```
Will do the same line that you post

Can send me next time a screenshot, if you have again this bug?
I would like to fix, but I can't reproduce otherwise.

Philliec459, 2020-06-03:
Thank you. 

I think the problem is that I have added a SSD and everything is now on SSD exept for what is needed to boot from micro SD card.

In the process on the old SD was a swfile swapfile. jtop was looking at that and not the swapfile on the SSD. Once deleting the swfile on the SD jtop is behaving properly. However, if I do create an Extra Swap with jtop, will jtop be looking just at the SD card for a swap file or can we direct it to look at the new SSD card?

One other question. I love having access to the fan speed in jtop especially while training a model. Sometimes at the end of training the fan will automatically switch to 0. I am glad I have jtop to observe this, but is there a way to have the fan speed adhere to what we have in jtop? 

Best Regards, 
 
_______________________________
 
E. Craig Phillips
CEO and Chief Petrophysicist
Crested Butte Petrophysical Consultants
 

459 Cisneros Lane
Crested Butte, CO 81224
USA
Office: +1 970-343-0730
Mobile: +1 970-343-0730
email: craig@cbpetro.com
Website: www.cbpetro.com
GitHub: https://github.com/Philliec459

The information contained in this electronic message is confidential, it constitutes a profesional and/or industrial secret in terms of the current legislation, and is intended for its recipient only. If you receive this message by mistake or if you are not the recipient thereof, please notify the sender and destroy it.

La informacion contenida en este mensaje de datos es confidencial, constituye un secreto industrial y/o profesional en terminos de la legislacion vigente y se encuentra dirigida exclusivamente al destinatario indicado en dicho mensaje. Si usted recibe esta informacion por error o si usted no es el destinatario del mensaje, favor de notificar al emisor, y destruyalo.

Philliec459, 2020-06-03:
I have now used jtop to add an additional 2GB of swap space and it is called swfile. jtop configured it for the new SSD (vs. the micro SD card that I still need to boot from). I used page 4MEM and clicked on 's' and it did disable the swap file and gave me the proper display to re-Enable swap and set the size of the swap file. This all working properly on the new SSD too. Thank you.

rbonghi, 2020-06-03:
> In the process on the old SD was a swfile swapfile. jtop was looking at that and not the swapfile on the SSD. Once deleting the swfile on the SD jtop is behaving properly. However, if I do create an [...]

I need to make a setup to test it. I can't reply right now, but I will check

> One other question. I love having access to the fan speed in jtop especially while training a model. Sometimes at the end of training the fan will automatically switch to 0. I am glad I have jtop to observe this, but is there a way to have the fan speed adhere to what we have in jtop?

I think depends if jetson_clocks is disabled from another script

Philliec459, 2020-06-03:
Thank you for your reply. Jtop and the swap file are all working fine now that I removed the swfile from the micro SD card and now jtop’s swfile is on the SSD. It should be faster that way too.

I think you are right about jetson_clocks. During the etreme training the CPUs are running at 1.4Hz and at the end go back to 1.2Hz. This is when the fan turns off. 

Best Regards, 
 
_______________________________
 
E. Craig Phillips
CEO and Chief Petrophysicist
Crested Butte Petrophysical Consultants
 

459 Cisneros Lane
Crested Butte, CO 81224
USA
Office: +1 970-343-0730
Mobile: +1 970-343-0730
email: craig@cbpetro.com
Website: www.cbpetro.com
GitHub: https://github.com/Philliec459

The information contained in this electronic message is confidential, it constitutes a profesional and/or industrial secret in terms of the current legislation, and is intended for its recipient only. If you receive this message by mistake or if you are not the recipient thereof, please notify the sender and destroy it.

La informacion contenida en este mensaje de datos es confidencial, constituye un secreto industrial y/o profesional en terminos de la legislacion vigente y se encuentra dirigida exclusivamente al destinatario indicado en dicho mensaje. Si usted recibe esta informacion por error o si usted no es el destinatario del mensaje, favor de notificar al emisor, y destruyalo.

closed 2020-06-03 by rbonghi