rbonghi/jetson_stats #66: "jetson_swap does not exist!" when swapfile is used

marcos69, 2020-06-30:
jtop version 2.1.0 does not start, and gives following error:
  [ERR] jetson_swap does not exist!
Run jetson_config (health page) to fix it...

But system have 16GB swapfile and 8x zram(0 to7)  each 2GB working, (listed in version 2.0.4)

**To Reproduce**
sudo jtop

**Expected behavior**
Just start, like the previous version: 2.0.4

**Board** Jetson Xavier
 - SOC:  tegra194
 - ID: 25
 - Board(s):  P2888-0001 P2822-0000
 - Code Name: galen

**Jetpack**
 - Jetpack: 4.4
 - L4T: 32.4.2

**Jetson-Stats**
 - Version: 2.1.0

rbonghi, 2020-06-30:
Hi @marcos69 

Look like jetson-stats is bad installed. Try to run these commands:
```
sudo -H pip uninstall jetson_stats
sudo -H pip install --no-cache-dir -U jetson_stats
```

Meanwhile I will check the reasons

marcos69, 2020-07-06:
ok it worked, thanks

closed 2020-07-06 by marcos69