rbonghi/jetson_stats #72: jetson_stas service failing on boot after 3.0 upgrade

lucasilvassis, 2020-07-21:
Hello!

We are experiencing a strange behavior after an upgrade to 3.0 version. After the upgrade, the jetson_stats service is having some problems initializing on boot.

The output of  **journalctl -u jetson_stats.service** is attached. Note that the boot is initializing at 15:39, with the service failing to initialize. Shortly after, at 15:41, we restart the service manually and it succeeds.

```
jul 21 15:39:53 xavier systemd[1]: Started jetson_stats service.
jul 21 15:39:54 xavier jtop[5056]: [INFO] jtop.core.config - Load config from /usr/local/jetson_stats/config.json
jul 21 15:39:54 xavier jtop[5056]: [INFO] jtop.core.common - fan loaded on /sys/devices/pwm-fan/
jul 21 15:39:54 xavier jtop[5056]: [INFO] jtop.core.common - jetson_clocks loaded on /usr/bin/jetson_clocks
jul 21 15:39:55 xavier jtop[5056]: [INFO] jtop.core.common - tegrastats loaded on /usr/bin/tegrastats
jul 21 15:39:55 xavier jtop[5056]: [INFO] jtop._main_ - jetson_stats server loaded
jul 21 15:39:55 xavier jtop[5056]: Process JtopServer-1:
jul 21 15:39:55 xavier jtop[5056]: Traceback (most recent call last):
jul 21 15:39:55 xavier jtop[5056]:   File "/usr/lib/python2.7/multiprocessing/process.py", line 267, in _bootstrap
jul 21 15:39:55 xavier jtop[5056]:     self.run()
jul 21 15:39:55 xavier jtop[5056]:   File "/usr/local/lib/python2.7/dist-packages/jtop/service.py", line 185, in run
jul 21 15:39:55 xavier jtop[5056]:     self.jetson_clocks.initialization(self.nvpmodel)
jul 21 15:39:55 xavier jtop[5056]:   File "/usr/local/lib/python2.7/dist-packages/jtop/core/jetson_clocks.py", line 243, in initialization
jul 21 15:39:55 xavier jtop[5056]:     lines = cmd(timeout=COMMAND_TIMEOUT)
jul 21 15:39:55 xavier jtop[5056]:   File "/usr/local/lib/python2.7/dist-packages/jtop/core/command.py", line 107, in _call_
jul 21 15:39:55 xavier jtop[5056]:     raise Command.TimeoutException('Error process:', self.process.returncode)
jul 21 15:39:55 xavier jtop[5056]: TimeoutException: [errno:1] Error process:
jul 21 15:39:56 xavier jtop[5056]: [INFO] jtop.service - Service closed
jul 21 15:41:38 xavier systemd[1]: Started jetson_stats service.
jul 21 15:41:39 xavier jtop[7952]: [INFO] jtop.core.config - Load config from /usr/local/jetson_stats/config.json
jul 21 15:41:39 xavier jtop[7952]: [INFO] jtop.core.common - fan loaded on /sys/devices/pwm-fan/
jul 21 15:41:39 xavier jtop[7952]: [INFO] jtop.core.common - jetson_clocks loaded on /usr/bin/jetson_clocks
jul 21 15:41:39 xavier jtop[7952]: [INFO] jtop.core.common - tegrastats loaded on /usr/bin/tegrastats
jul 21 15:41:39 xavier jtop[7952]: [INFO] jtop._main_ - jetson_stats server loaded
jul 21 15:41:40 xavier jtop[7952]: [INFO] jtop.core.fan - Mode set default status=False
jul 21 15:41:40 xavier jtop[7952]: [INFO] jtop.core.fan - Mode set default status=True
jul 21 15:41:40 xavier jtop[7952]: [INFO] jtop.core.jetson_clocks - jetson_clocks started
```

Can you help us sort the  Command.TimeoutException throw, as well as guide us through possible solutions?

rbonghi, 2020-07-21:
Hi @lucasilvassis ,

Thank you for your issue! I will check on my boards! I notice that you are using a Xavier, which one? AGX Xavier or Xavier NX?

I see that the jetson_stats.service restart and look like works after the TimeoutException. Is it true?

Looks like jetson_clocks crashed or does not replied in time in the beginning.

lucasilvassis, 2020-07-21:
We are using a Jetson Xavier with Jetpack 4.3, sorry for not mentioning before.

Exactly, the service, as well as jtop, are working after the manual restart, the issue only occurs on boot.

rbonghi, 2020-07-21:
I am sorry to ask you a lot of questions :-)

This error occur for every reboot or is occasionally?
I understanding which type of bug is it and try to make a similar setup to test with my board. Luckily we have the same board with same JP :-D

lucasilvassis, 2020-07-21:
Awesome! Well, we experienced this behavior on every boot so far, It may be a permanent thing.

I'll try to bring a little more context to the circumstances of this issue. We have some other services running (VPN stuff, not related to jetson_clocks) on boot that are working as intended. We were suspecting some kind of throttling at start and tried to raise TimeoutStartSec and even the COMMAND_TIMEOUT mentioned in the error, however, we had no success yet.

rbonghi, 2020-07-21:
Thank you @lucasilvassis !

I will try the same on my board, overall I think is something manageable. I hope will not require enough time to fix it.
Are you working only with jtop gui or are you working with jtop API?

About the exception, you can only find in my code: https://github.com/rbonghi/jetson_stats/blob/master/jtop/core/command.py#L107
I raised an exception and I don't catch it :laughing:

lucasilvassis, 2020-07-21:
Awesome! We'll also update here if we find any solution.

We just started working with the API but we mainly use the gui for now.

closed 2020-07-25 by rbonghi