rbonghi/jetson_stats #76: Temperature rise and memory of the jetson nano after long use

yerayb, 2020-08-06:
Hi, I'm using jetson stats in a python script. The python script has a while that constantly sends Jetson values to a DB. The problem is that after a few hours, the ram, cpu and temperature increase considerably. What can I do to stop this from happening? 

`def datos_mensaje():
    
    configuracion_aws=leer_parametros_configuración()

    with jtop() as jetson:
        payload = {
            "Timestamp" : str(datetime.now()), #Fecha y hora actual 
            "Maquina": configuracion_aws['cliente_jetson'],
            "CPU1": str(jetson.cpu['CPU1']['val']) ,
            "CPU2": str(jetson.cpu['CPU2']['val']) ,
            "CPU3": str(jetson.cpu['CPU3']['val']) ,
            "CPU4": str(jetson.cpu['CPU4']['val']) ,
            "GPU": str(jetson.gpu['val']),
            "RAMconsum": str(jetson.ram['use']) ,
            "RAMtotal": str(jetson.ram['tot']) ,
            "Temp PPl": str(jetson.temperature['PLL']) ,
            "Temp CPU": str(jetson.temperature['CPU']) ,
            "Temp GPU": str(jetson.temperature['GPU']) ,
            "Thermal": str(jetson.temperature['thermal']) ,
            "Temp AO": str(jetson.temperature['AO']) 
          }

    return json.dumps(payload) #Convierte string a JSON`

I call this function from the main with a while True with a time.sleep of 30 seconds.

Thanks

rbonghi, 2020-08-06:
Hi @yerayb ,

Thank you for open this issue :-)

I notice that you forget to use [jetson.ok()](https://rbonghi.github.io/jetson_stats/jtop.html#jtop.jtop.jtop.ok) this blocking function unlock the rest of the function only when a new data is available, avoiding long loop when it is send the same data.

Try this option:
```python
with jtop() as jetson:
  while jetson.ok():
    payload = {
        "Timestamp" : str(datetime.now()), #Fecha y hora actual 
        "Maquina": configuracion_aws['cliente_jetson'],
        "CPU1": str(jetson.cpu['CPU1']['val']) ,
        "CPU2": str(jetson.cpu['CPU2']['val']) ,
        "CPU3": str(jetson.cpu['CPU3']['val']) ,
        "CPU4": str(jetson.cpu['CPU4']['val']) ,
        "GPU": str(jetson.gpu['val']),
        "RAMconsum": str(jetson.ram['use']) ,
        "RAMtotal": str(jetson.ram['tot']) ,
        "Temp PPl": str(jetson.temperature['PLL']) ,
        "Temp CPU": str(jetson.temperature['CPU']) ,
        "Temp GPU": str(jetson.temperature['GPU']) ,
        "Thermal": str(jetson.temperature['thermal']) ,
        "Temp AO": str(jetson.temperature['AO']) 
      }
      return json.dumps(payload)
```

If you are running this script from another function, I suggest to setup in another way:

```python
from jtop import jtop

jetson = jtop()
jetson.start()

def parse_data(jetson):
    ... YOUR PARSING...
    return json.dumps(payload)

while jetson.ok():
    ... YOUR CODE ...
    payload = parse_data(jetson)
    ... YOUR CODE ...

jetson.close()
```

Last alternative you can use a callback

```python
from jtop import jtop

payload = {}

def read_stats(jetson):
    global payload
    ... YOUR CODE ...
   payload = ...

jetson = jtop()
jetson.attach(read_stats)
jetson.start()

while True:
    ... YOUR CODE ...
    data = json.dumps(payload)
    ... YOUR CODE ...

jetson.close()
```

Keep me posted if you fixed

yerayb, 2020-08-06:
Thank you very much for your response, the last alternative using the callback was the solution I needed. The values are almost not altered and my jetson does not heat up as it used to. Thank you!

closed 2020-08-06 by yerayb