rbonghi/jetson_stats #77: nvdc: nvdcEventThread select failed: Bad file descriptor

loibucket, 2020-08-12:
The 3.0.1 version makes an error message "nvdc: nvdcEventThread select failed: Bad file descriptor" when starting a new terminal tab in the jupyter lab.  It does not affect operability as the terminal window still works, but there's just this error message when it starts.  Reverting to 3.0.0 seems to have fixed this issue.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...' jupyter lab on another pc
2. Click on '....' launcher -> new terminal
...
4. See error "nvdc: nvdcEventThread select failed: Bad file descriptor"

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Add any other context about the problem here.

**Board**
 - SOC: [e.g. tegraXXX]
 - ID: [e.g. 10]
 - Board(s): [e.g. P1000-1000]
 - Code Name: [e.g. test]

**Jetpack**
 - Jetpack: [e.g. 4.3] 4.3

**Jetson-Stats**
 - Version: [e.g. 1.8] 3.0.1

rbonghi, 2020-08-13:
Hi @loibucket ,

Thank you for your message, really clear.
I honestly never tried jtop on *jupyter lab* I will check it and I will keep you posted.

Have you tried jtop outside *jupyter lab*?

loibucket, 2020-08-15:
Yes, jtop works fine outside jupyter lab.  It also works fine inside jupyter lab's terminal.  The error message just shows up as the first line when a new terminal window is opened inside jupyter lab.  It does not affect the functionality of jtop.

rbonghi, 2020-08-16:
Hi @loibucket ,

I suggest to write a new post on NVIDIA forum: https://forums.developer.nvidia.com/ 
Watching on google is something on Vulkan

closed 2020-08-16 by rbonghi