rbonghi/jetson_stats #80: Uuse jtop in python script inside docker container failling

assafzam, 2020-09-02:
Hi,
 
I've installed jetson-stats successfully inside a docker container via:

`root@office:/var/remote-python# sudo -H pip install -U jetson-stats`

and than run the exapmle from your docs:
```
root@office:/var/remote-python# python

Python 3.6.12 (default, Sep  1 2020, 14:19:07)
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.`

>>> from jtop import jtop
>>> with jtop() as jetson:
...     # jetson.ok() will provide the proper update frequency
...     while jetson.ok():
...         # Read tegra stats
...         print(jetson.stats)
```
And got:
```
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/jtop/jtop.py", line 927, in start
    self._broadcaster.connect()
  File "/usr/local/lib/python3.6/multiprocessing/managers.py", line 489, in connect
    conn = Client(self._address, authkey=self._authkey)
  File "/usr/local/lib/python3.6/multiprocessing/connection.py", line 487, in Client
    c = SocketClient(address)
  File "/usr/local/lib/python3.6/multiprocessing/connection.py", line 614, in SocketClient
    s.connect(address)
FileNotFoundError: [Errno 2] No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/site-packages/jtop/jtop.py", line 1081, in __enter__
    self.start()
  File "/usr/local/lib/python3.6/site-packages/jtop/jtop.py", line 930, in start
    raise JtopException("The jetson_stats.service is not active. Please run:\nsudo systemctl restart jetson_stats.service")
jtop.core.exceptions.JtopException: The jetson_stats.service is not active. Please run:
sudo systemctl restart jetson_stats.service
```

what can I do to fix it? 
(cannot run 'sudo systemctl restart jetson_stats.service' inside the container in automation pipeline)

rbonghi, 2020-09-06:
I test your docker and the build works, remember when you run your container to add the volume: `/run/jtop.sock:/run/jtop.sock`

like this example:
`docker run --rm -it -v /run/jtop.sock:/run/jtop.sock rbonghi/jetson-stats`

This Dockerfile works fine for me:
```dockerfile
FROM arm64v8/python:3-buster
RUN pip install -U jetson-stats
```

**Hints:**
* _sudo_ is not allowed in a Dockerfile, I suggest to don't install packages with sudo
* Use of apt-get update should be paired with rm -rf /var/lib/apt/lists/* in the same layer.
* Consider using a --no-install-recommends when apt-get installing packages.

Other hints are here: https://docs.docker.com/develop/develop-images/dockerfile_best-practices/

assafzam, 2020-09-06:
@rbonghi great! Thanks

closed 2020-10-28 by rbonghi