rbonghi/jetson_stats #82: top says usage is <1%, I start jtop and then usage is at 100%...

AlexisWilke, 2020-09-07:
**Is your feature request related to a problem? Please describe.**
A tool such as `top` is expected to have a minimal impact on the CPUs otherwise it's really difficult to know what uses the CPU(s).

The current version uses between 5% and 25% of each of my 4 CPUs. Most often around a total of 100% CPU time.

**Describe the solution you'd like**
The tool should somehow be faster. For example, `top` uses _just_ 0.7% to 1%.

One thing `top` does is redraw only what's necessary in an attempt to reduce the time to send characters to the screen as much as possible.

**Describe alternatives you've considered**
For now I use `top` and `sudo tegrastats` which give me the data I'm interested in although not as good as `jtop`, these tools are fast.

**Additional context**
I run on a Jetson AGX Xavier.

Also, I run remotely (but I would imagine many of us do). Since `top` is not affected by that part, I would think that `jtop` wouldn't either.

Note that the `jetson_stats` does not seem to use much CPU at all. Under 1% most of the time.

rbonghi, 2020-09-07:
Thank you for your message and detail about jetson_stats. This is a know issue #73 ( and other issues that I do not remember).
The main reason now is the use of sub-processes to read the status of the Jetson board.
At this time jtop run different supbrocess in background to print the status:
* tegrastats
* jetson_clocks
* /proc/meminfo
* fan

but the heavy processes that I notice are **tegrastats** and **jetson_clocks**.
It's in a plan for the next revision to rewrite that parts directly reading from the system the status of the board.
I expect to reduce the usage of jtop for another 10-15% but I do not think the status of jtop will be less than *top*
I do not have right now an experience to improve the performance of jtop in a python script, but I am working on that.

AlexisWilke, 2020-09-07:
Ah. I see that #73 is closed, which would be why I did not see it when for looking to make sure I don't open a duplicate.

Yes. Directly reading the files could be much faster. I did not realize that you ran `tegrastats` to get that info. In top/htop, though, `tegrastats` does not appear to use much CPU. On the other hand, I can see `jetson_clocks` quite a bit. Well... I hope you can reduce the the usage. Right now it's great to tweak a few things and check the NVDEC and such parameters, but CPU wise, it's not practical while running my own app.