rbonghi/jetson_stats #83: Disecting the installation command

muzammil360, 2020-09-11:
**Describe the bug**
Hi @rbonghi, I wanted to understand disect the command you share for installation 

`
sudo -H pip install -U jetson-stats
`

From the documentation, it seems that `-U` will **Upgrade all specified packages to the newest**. It means that this flag will only make sure you always have the latest packages installed after running it. 

However, `-H` is a bit interesting. `sudo` documentation says that it will set the `HOME` directory to current user's directory instead of `/root`. In this case, will the `jetson-stats` package be installed in system-wide location `/usr/local/lib/python3.6/dist-packages` or in `${HOME}/.local/lib/python3.6/site-packages`?

rbonghi, 2020-10-25:
Hi @muzammil360 ,

I am sorry for my late reply, you have right. I start to use `-H` in the beginning but now is not necessary.
I will check on my board, just to be sure all work and I will remove from all readme.

Thank you!

rbonghi, 2023-01-12:
Picking up an old issue:

Install without -H, I catch always this warning:

> WARNING: The directory '/home/nvidia/.cache/pip' or its parent directory is not owned or is not writable by the current user. The cache has been disabled. Check the permissions and owner of that directory. If executing pip with sudo, you should use sudo's -H flag.

I need to investigate

rbonghi, 2023-01-20:
Following https://askubuntu.com/questions/744377/cache-pip-disabled-immediately-after-instalation-of-pip

I remove this option from next release

closed 2023-01-25 by rbonghi