rbonghi/jetson_stats #84: Authentication mismatch with jetson-stats server

CNSaintNeo, 2020-09-16:
When I run jtop after being installed，it rise the words that authentication mismatch with jetson-stats server without any hints.
How can we go through this problem， thanks！

cylee81, 2020-10-13:
Reboot the board is helpful!

rbonghi, 2020-10-25:
Hi @CNSaintNeo 
I am sorry for my late reply, but if you fixed following the @cylee81 hint this issue should be close. Otherwise, keep free to open again this issue

closed 2020-10-25 by rbonghi

tensorturtle, 2021-04-01:
I encountered this issue when trying to use `jtop_logger.py` it in my project. The silly mistake was using Python 3 to run the script instead of Python 2.7.

For example, I should NOT have done `python3 jtop_logger.py`

and instead use: `python jtop_logger.py`.