rbonghi/jetson_stats #85: Power Avg

assafzam, 2020-09-16:
Hi @rbonghi, 

can you please explain what is the `power avg`  attribute in `jtop.stats`, I mean Average of what?
Thanks!

dx2919717227, 2020-09-24:
#64

assafzam, 2020-09-24:
> #64

Thanks, I saw that.  
I meant in my question, Average on what values? (from last half hour? last 5 minuets?)

dx2919717227, 2020-09-24:
Average = (power)/(time)   
The time is counted from the moment you run the command.（seconds）
The power is the total energy consumption during this time.（mW）

kmenezes, 2020-09-29:
Hello, in the power monitoring section is the ALL indicative of total board power? The CPU GPU CV rails plus the SOC rail don't seem to add up with what is reported with tegrastats. What is the best way to capture total board power with the application?

rbonghi, 2020-10-25:
Hi @assafzam ,

The informations from **Power avg** are read from **tegrastats**

https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%20Linux%20Driver%20Package%20Development%20Guide/power_management_nano.html

closed 2020-10-25 by rbonghi