rbonghi/jetson_stats #86: jetson_stats.service: inactive

pangyuhang, 2020-09-30:
**Describe the bug**
- NVIDIA Jetson AGX Xavier [16GB]
   * Jetpack 4.3 [L4T 32.3.1]
   * NV Power Mode: MAXN - Type: 0
   * jetson_stats.service: inactive
I'm confused about * jetson_stats.service: inactive. 
While running jtop in terminal, the following warnings occur:
**To Reproduce**
Steps to reproduce the behavior:
1. Terminal: 
agx1@agx1-desktop:~$ jtop
The jetson_stats.service is not active. Please run:
sudo systemctl restart jetson_stats.service
2. agx1@agx1-desktop:~$ sudo systemctl restart jetson_stats.service
Failed to restart jetson_stats.service: Unit jetson_stats.service not found.

rbonghi, 2020-10-25:
Hi @pangyuhang 

Have you tried to restart the board?

awesome-doge, 2020-11-05:
how to restart the board?
do you mean reboot the computer?

rbonghi, 2020-11-05:
Hi @awesome-doge ,

yes, reboot your board.
Otherwise, try to reinstall jetson-stats follow this command:
```
sudo -H pip install --no-cache-dir -U jetson-stats
```

asoong98, 2020-12-31:
Worked for me. Need to make sure to pip3 uninstall jetson-stats and reboot the board, then proceed with the reinstallation.

Adrian-Yan16, 2021-09-26:
I reinstall jetson-stats, but it show the same error, while i run the command "sudo jtop", it works.

JNaranjo-Alcazar, 2023-01-03:
Hi, I am trying to run this using the docker image on Orin and I get the following error:

```
docker run -v /run/jtop.sock:/run/jtop.sock --name jetson_stats rbonghi/jetson_stats:master 
The jetson_stats.service is not active. Please run:
sudo systemctl restart jetson_stats.service
```

rbonghi, 2023-01-03:
Hi @JNaranjo-Alcazar I'm still fixing these bugs

JNaranjo-Alcazar, 2023-01-03:
Okey @rbonghi , thanks for the quick reply!

rbonghi, 2023-01-03:
Hi @JNaranjo-Alcazar and @Adrian-Yan16 ,

can you send me the log coming from:
```
journalctl -u jetson_stats.service
```

JNaranjo-Alcazar, 2023-01-03:
Okey, I will send it to you tomorrow
Sorry for the delay

rbonghi, 2023-01-05:
Do you know if there is any news?

Meanwhile, I made a new release candidate:
```
sudo -H pip install jetson-stats==4.0.0rc3
````

JNaranjo-Alcazar, 2023-01-05:
Hi, I was going to send you my output today. I hope I can send you before 2pm

rbonghi, 2023-01-05:
That's great!
Could you try on my new release candidate?
I have the feeling that docker was not connected with the latest jetson_stats version.

JNaranjo-Alcazar, 2023-01-05:
```
docker run -it -v /run/jtop.sock:/run/jtop.sock --name jetson_stats python:3-buster bash
```
Inside the docker container:

```
pip install jetson-stats==4.0.0rc3
jtop
The jetson_stats.service is not active. Please run:
sudo systemctl restart jetson_stats.service
```
Should I try something different?

rbonghi, 2023-01-05:
Some questions:
Could you install from your host the same jetson-stats version?
Do you know if your host jtop works?

JNaranjo-Alcazar, 2023-01-05:
I have not tried to install jetson-stats in the host.
The idea is to use docker to run python packges (pip).

rbonghi, 2023-01-05:
Yes, you need jetson-stats to install to your host also.

When jtop is running on your container, you share a volume: `/run/jtop.sock:/run/jtop.sock` that make able a jtop to communicate with the master

JNaranjo-Alcazar, 2023-01-05:
Okey, I am going to install it

JNaranjo-Alcazar, 2023-01-05:
Hi, now jtop work on the host.
When running 
```
docker run -v /run/jtop.sock:/run/jtop.sock rbonghi/jetson_stats:4.0.0rc3
Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/local/lib/python3.11/site-packages/jtop/__main__.py", line 147, in main
    curses.wrapper(JTOPGUI, jetson, [ALL, GPU, CPU, MEM, CTRL, INFO], init_page=args.page, loop=args.loop, seconds=LOOP_SECONDS)
  File "/usr/local/lib/python3.11/curses/__init__.py", line 73, in wrapper
    stdscr = initscr()
             ^^^^^^^^^
  File "/usr/local/lib/python3.11/curses/__init__.py", line 29, in initscr
    setupterm(term=_os.environ.get("TERM", "unknown"),
_curses.error: setupterm: could not find terminal
```

rbonghi, 2023-01-05:
It's missing the command `-t` or `--tty`
https://docs.docker.com/engine/reference/commandline/run/
In this case, docker cannot run jtop because there are no terminals. If you try:

```
docker run -v /run/jtop.sock:/run/jtop.sock -t rbonghi/jetson_stats:4.0.0rc3
```
You will see jtop running inside the terminal

Note: the option `-t` is not needed if you are running a python script that uses the jtop python library

JNaranjo-Alcazar, 2023-01-05:
```
docker run -v /run/jtop.sock:/run/jtop.sock -t rbonghi/jetson_stats:4.0.0rc3
Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/local/lib/python3.11/site-packages/jtop/__main__.py", line 147, in main
    curses.wrapper(JTOPGUI, jetson, [ALL, GPU, CPU, MEM, CTRL, INFO], init_page=args.page, loop=args.loop, seconds=LOOP_SECONDS)
  File "/usr/local/lib/python3.11/curses/__init__.py", line 94, in wrapper
    return func(stdscr, *args, **kwds)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/jtop/gui/jtopgui.py", line 109, in __init__
    self.run(loop, seconds)
  File "/usr/local/lib/python3.11/site-packages/jtop/gui/jtopgui.py", line 138, in run
    self.draw()
  File "/usr/local/lib/python3.11/site-packages/jtop/gui/lib/common.py", line 80, in wrapped
    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/jtop/gui/jtopgui.py", line 149, in draw
    self.header()
  File "/usr/local/lib/python3.11/site-packages/jtop/gui/lib/common.py", line 120, in wrapped
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/jtop/gui/jtopgui.py", line 186, in header
    status = [self.jetson.board.info["model"]]
              ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^
KeyError: 'model'

```

rbonghi, 2023-01-05:
Do you know if your host jtop works?
if true, it's using the same jetson-stats version?

JNaranjo-Alcazar, 2023-01-05:
yes, in host works and I am using (3.1.4). I do not know the version installed in the image rbonghi/jetson_stats:4.0.0rc3

rbonghi, 2023-01-05:
The new 4.0.0 version has deep changes compared to 3.1.4 and this is the reason you see this failure.

I suggest updating also your host with the 4.0.0rc3 or wait this afternoon and updating all versions (host and docker) with the latest version

JNaranjo-Alcazar, 2023-01-05:
It is taking a long time to install 4.0.0rc3. Is it normal?

rbonghi, 2023-01-05:
can happens, because shutdown a service and reboot.
If is stuck stop the installation and retry, should work

closed 2023-01-05 by rbonghi

mmldxxn, 2023-05-31:
Hi,
I am also tring to install jetson-stats, but every time i stuck at this 

 **sudo -H pip3 install -U jetson-stats
Collecting jetson-stats
  Using cached https://files.pythonhosted.org/packages/aa/07/098bfb6f864b44c12957be8798c34af4faabac33ffde2eaf1ef861f901e5/jetson-stats-4.2.1.tar.gz
Requirement already up-to-date: distro in /usr/local/lib/python3.6/dist-packages (from jetson-stats)
Requirement already up-to-date: smbus2 in /usr/local/lib/python3.6/dist-packages (from jetson-stats)
Building wheels for collected packages: jetson-stats
  Running setup.py bdist_wheel for jetson-stats ... |_**

for a long time. I have tried for several times, but every time it is same. Can anyone help me?

commit abea13db (2023-01-05): Improve readme, fix #86 and version change