rbonghi/jetson_stats #87: Board missing: Xavier 32GB

domcross, 2020-09-30:
Please update jetson-stats with this board

**Board**
   - Type: AGX Xavier [16GB]
   - SOC Family: tegra194 - ID:25
   - Module: P2888-0001 - Board: P2822-0000
   - Code Name: galen

**Jetpack**
 - Jetpack: 4.4
 - L4T: 32.4.3

**Jetson-Stats**
 - Version: 3.0.1

My board is Xavier with 32GB RAM but jtop 3.0.1 shows it as "Xavier 16GB" (red markers in attached screenshot). The memory bar shows correct amount though (yellow marker).
[image]

rbonghi, 2020-10-25:
Hi @domcross 

I notice the same bug as well, I need a modification on my jetson variable reader to fix it. After the release to fix #90 I will fix it

rbonghi, 2023-01-05:
This error should now be fixed with the latest release **4.0** try to install and check again
```
sudo -H pip install -U jetson-stats
```

closed 2023-01-05 by rbonghi