rbonghi/jetson_stats #88: setup.py

celikemir, 2020-10-07:
Hi, I have no internet connection on my Nvidia Jetson Nx board so I have tried to install jetson-stat using setup.py script with the following command `sudo python3 setup.py install` I  attached the error below. I haven't pip on my system so that please don't suggest pip usage. I have installed setuptools flawlessly with setup.py before. I also tried `sudo python3 setup.py develop` it works but jtop not.

`nvidia@nvidia:/usr/jetson_stats-master$ sudo python3 setup.py install
running install
Traceback (most recent call last):
  File "/usr/jetson_stats-master/jtop/jtop.py", line 927, in start
    self._broadcaster.connect()
  File "/usr/lib/python3.6/multiprocessing/managers.py", line 489, in connect
    conn = Client(self._address, authkey=self._authkey)
  File "/usr/lib/python3.6/multiprocessing/connection.py", line 487, in Client
    c = SocketClient(address)
  File "/usr/lib/python3.6/multiprocessing/connection.py", line 614, in SocketClient
    s.connect(address)
FileNotFoundError: [Errno 2] No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 33, in <module>
    sys.exit(load_entry_point('jetson-stats', 'console_scripts', 'jtop')())
  File "/usr/jetson_stats-master/jtop/__main__.py", line 131, in main
    with jtop(interval=interval) as jetson:
  File "/usr/jetson_stats-master/jtop/jtop.py", line 1081, in __enter__
    self.start()
  File "/usr/jetson_stats-master/jtop/jtop.py", line 930, in start
    raise JtopException("The jetson_stats.service is not active. Please run:\nsudo systemctl restart jetson_stats.service")
jtop.core.exceptions.JtopException: The jetson_stats.service is not active. Please run:
sudo systemctl restart jetson_stats.service
[Uninstall commands - Check all commands if are installed]
Nothing to do
[Uninstall folders - Check folders]
Nothing to do
[Uninstall variables - Check variables status]
No jetson environment variables scripts installed
[Uninstall permissions - Check status permissions for nvidia]
Nothing to do
[Uninstall service - Status jetson-stats service]
System prefix /usr
error: [Errno 2] No such file or directory: '/services/jetson_env.sh'
`

celikemir, 2020-10-07:
I solved myself,

- `sudo python3 setup.py develop`

- copy under service folder to `/usr/local/jetson-stats` manually

- install importlib_metadata and  zipp-3.3.0  from source code.

setup.py works wrong on my nx , I have no virtualenv on it but its detect virtualenv.

rbonghi, 2020-10-25:
Hi @celikemir ,

Usually, when I develop on jetson-stats I run the command below (from the jetson-stats folder)
```
sudo -H pip install -e .
```
that should be the same of `sudo python3 setup.py develop`

I will check manage a way to install properly with `sudo python3 setup.py install` for offline installation.

About **importlib_metadata** and **zipp-3.3.0** I will check on my jetson, theoretically should not require any other package.

About virtualenv right now does not work, but I will release a new version of jetson-stats that will be permitted to install, now follow the issue #56

closed 2022-06-13 by rbonghi