rbonghi/jetson_stats #90: Jetpack missing [L4T 32.4.4]

JoeyWu88, 2020-10-22:
Please update jetson-stats with new jetpack

**Linux for Tegra**
 - L4T: 32.4.4

**Jetson-Stats**
 - Version: 3.0.1

rbonghi, 2020-10-24:
Thank you @JoeyWu88 
It will be fixed soon

Qengineering, 2020-10-26:
JetPack **4.4.1** not recognized in L4T 32.4.4
[image]

rbonghi, 2020-10-28:
Fix with 63bb30e

Please update your jetson-stats with:
```
sudo -H pip install -U jetson-stats
```

closed 2020-10-28 by rbonghi

Qengineering, 2020-10-28:
Sorry Raffaello, no improvement.
[image]

rbonghi, 2020-10-28:
try to run this command:
```
sudo systemctl restart jetson_stats.service
```
or reboot your board

Keep me updated

Qengineering, 2020-10-28:
`systemctl restart` did the job! Thanks