rbonghi/jetson_stats #101: jtop stop working in Jetson Xavier NX, jetpack 4.4.1

FPSychotic, 2020-10-29:
it was working , but probably got broken after uninstall other's software dependency. Last thing I did was install opencv 3.4.6 which uninstalled the opencv3.2 that comes with ROS. I have sourced a catkin isolated, can be it is looking by the dependency there? 

the software everytime is better, I feel jelous. hrrr
take care mate

 
`Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.6/dist-packages/jtop/__main__.py", line 146, in main
    curses.wrapper(JTOPGUI, jetson, [ALL, GPU, CPU, MEM, CTRL, INFO], init_page=args.page, loop=args.loop, seconds=LOOP_SECONDS)
  File "/usr/lib/python3.6/curses/__init__.py", line 94, in wrapper
    return func(stdscr, *args, **kwds)
  File "/usr/local/lib/python3.6/dist-packages/jtop/gui/jtopgui.py", line 109, in __init__
    self.run(loop, seconds)
  File "/usr/local/lib/python3.6/dist-packages/jtop/gui/jtopgui.py", line 136, in run
    while not self.events() and self.jetson.ok(spin=True):
  File "/usr/local/lib/python3.6/dist-packages/jtop/jtop.py", line 1056, in ok
    raise ex_value
  File "/usr/local/lib/python3.6/dist-packages/jtop/jtop.py", line 149, in _load_jetson_libraries
    libraries = import_os_variables(JTOP_FOLDER + "/jetson_libraries", "JETSON_")
  File "/usr/local/lib/python3.6/dist-packages/jtop/core/common.py", line 97, in import_os_variables
    value = tup[1].strip()
IndexError: list index out of range
`

js1972, 2021-02-14:
This is not working on the Jason nano either with exactly the same error.
The only thing I did was update deepstream from 5.0 to 5.0.1.

```
$ head -n 1 /etc/nv_tegra_release

# R32 (release), REVISION: 4.3, GCID: 21589087, BOARD: t210ref, EABI: aarch64, DATE: Fri Jun 26 04:38:25 UTC 2020
```

rbonghi, 2023-01-05:
This error should now be fixed with the latest release **4.0** try to install and check again
```
sudo -H pip install -U jetson-stats
```

closed 2023-01-05 by rbonghi