rbonghi/jetson_stats #111: Is jtop display GPU usage while doing video decoding/encoding hardware acceleration?

HaifeiYang, 2020-11-14:
I know that GPU on Jetson use nvdec/nvenc to accelerate video encoding/decoding from 
[nvidia-video-codec-sdk](https://developer.nvidia.com/nvidia-video-codec-sdk)
But I want to know if jtop display GPU usage while doing video decoding/encoding hardware acceleration?
If not, how do I monitor the usage of nvdec/nvenc?
Thanks!

bharatik-IISc, 2021-08-02:
I have the same question.

rbonghi, 2023-01-05:
I don't have access for each process the engine use, but I can ready the shared memory used on Jetson.

[image]

The final release will be out at the end of this day. Keep ready to try it.
Meanwhile close this issue, if there are new suggestion feel free to reopen it

closed 2023-01-05 by rbonghi