rbonghi/jetson_stats #121: CUDA not installed?

neilyoung, 2021-01-01:
Hi,

`jetson release -v` delivers this on my Jetson Nano:

```
neil@jetson:~$ jetson_release -v
 - NVIDIA Jetson Nano (Developer Kit Version)
   * Jetpack 4.4.1 [L4T 32.4.4]
   * NV Power Mode: MAXN - Type: 0
   * jetson_stats.service: active
 - Board info:
   * Type: Nano (Developer Kit Version)
   * SOC Family: tegra210 - ID:33
   * Module: P3448-0000 - Board: P3449-0000
   * Code Name: porg
   * Boardids: 3448
   * CUDA GPU architecture (ARCH_BIN): 5.3
   * Serial Number: 1423119033466
 - Libraries:
   * CUDA: NOT_INSTALLED
   * cuDNN: 8.0.0.180
   * TensorRT: 7.1.3.0
   * Visionworks: 1.6.0.501
   * OpenCV: 4.1.1 compiled CUDA: NO
   * VPI: 0.4.4
   * Vulkan: 1.2.70
 - jetson-stats:
   * Version 3.0.2
   * Works on Python 3.6.9

```

The SD is created from the `Getting Started` section of Jetson Nano.

I'm now a bit concerned about the CUDA NO statement, since there _seems_ to be a CUDA installation on board:

```
neil@jetson:~$ /usr/local/cuda-10.2/bin/nvcc --version
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2019 NVIDIA Corporation
Built on Wed_Oct_23_21:14:42_PDT_2019
Cuda compilation tools, release 10.2, V10.2.89
neil@jetson:~$ 

```

What am I doing wrong?

neilyoung, 2021-01-01:
Solved

closed 2021-01-01 by neilyoung

TungTT1311, 2021-05-14:
> Solved

How did you fix it?

arun-kumark, 2022-09-25:
On my Jetson baord also, everything seems installed:
~/Documents/jetson-inference$ jetson_release -v
 - NVIDIA Jetson Xavier NX (Developer Kit Version)
   * Jetpack 4.6 [L4T 32.6.1]
   * NV Power Mode: MODE_20W_6CORE - Type: 8
   * jetson_stats.service: active
 - Board info:
   * Type: Xavier NX (Developer Kit Version)
   * SOC Family: tegra194 - ID:25
   * Module: P3668 - Board: P3509-000
   * Code Name: jakku
   * CUDA GPU architecture (ARCH_BIN): 7.2
   * Serial Number: 1425020057815
 - Libraries:
   * CUDA: 10.2.300
   * cuDNN: 8.2.1.32
   * TensorRT: 8.0.1.6
   * Visionworks: 1.6.0.501
   * OpenCV: 4.6.0 compiled CUDA: YES
   * VPI: ii libnvvpi1 1.1.15 arm64 NVIDIA Vision Programming Interface library
   * Vulkan: 1.2.70
 - jetson-stats:
   * Version 3.1.4
   * Works on Python 3.6.9
~/Documents/jetson-inference$

But when I run the container, it says 
./video-viewer: error while loading shared libraries: libnppicc.so.10: cannot open shared object file: No such file or directory

Looks to me its related to CUDA not installed.

Could you solve it or hint how it was solved for you ?

thanks