rbonghi/jetson_stats #123: Program jtop does not execute in UART terminal. Throws curses init_pair error.

anamud, 2021-01-14:
### Bug

The program jtop does not execute in a UART terminal. Exception is thrown from curses init_pair function.

```
$ jtop
Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.6/dist-packages/jtop/__main__.py", line 146, in main
    curses.wrapper(JTOPGUI, jetson, [ALL, GPU, CPU, MEM, CTRL, INFO], init_page=args.page, loop=args.loop, seconds=LOOP_SECONDS)
  File "/usr/lib/python3.6/curses/__init__.py", line 94, in wrapper
    return func(stdscr, *args, **kwds)
  File "/usr/local/lib/python3.6/dist-packages/jtop/gui/jtopgui.py", line 76, in __init__
    curses.init_pair(1, curses.COLOR_RED, curses.COLOR_BLACK)
_curses.error: init_pair() returned ERR
```

### Environment details

```
$ echo $TERM
vt220
$ jetson_release -v
 - NVIDIA Jetson Nano (Developer Kit Version)
   * Jetpack 4.4.1 [L4T 32.4.4]
   * NV Power Mode: MAXN - Type: 0
   * jetson_stats.service: active
 - Board info:
   * Type: Nano (Developer Kit Version)
   * SOC Family: tegra210 - ID:33
   * Module: P3448-0000 - Board: P3449-0000
   * Code Name: porg
   * Boardids: 3448
   * CUDA GPU architecture (ARCH_BIN): 5.3
   * Serial Number: 0421319028241
 - Libraries:
   * CUDA: NOT_INSTALLED
   * cuDNN: NOT_INSTALLED
   * TensorRT: NOT_INSTALLED
   * Visionworks: NOT_INSTALLED
   * OpenCV: NOT_INSTALLED compiled CUDA: NO
   * VPI: NOT_INSTALLED
   * Vulkan: 1.2.70
 - jetson-stats:
   * Version 3.0.2
   * Works on Python 3.6.9
```

anamud, 2021-01-14:
Additionally, I have ensured that these ncurses [tests](https://invisible-island.net/ncurses/ncurses-examples.html) execute on the UART terminal. They execute without color which is fine. I expect jtop to work without color support. Please let me know if this is not the case i.e., colors are essential for jtop working.

awandzel, 2021-10-06:
Same problem here for headless display

awandzel, 2021-10-06:
Fixed it by setting: `export TERM='xterm-256color'`, which sets your terminal to have default 256 color.

rbonghi, 2023-01-05:
This error should now be fixed with the latest release **4.0** try to install and check again
```
sudo -H pip install -U jetson-stats
```

closed 2023-01-05 by rbonghi