rbonghi/jetson_stats #124: FR: GPU Metrics Prometheus Exporter

danmx, 2021-01-21:
**Is your feature request related to a problem? Please describe.**

I'd like to scrape GPU metrics that expose `jtop` using [Prometheus](https://prometheus.io/)

**Describe the solution you'd like**

`jtop` or other tool would expose [a metric API](https://github.com/prometheus/client_python#prometheus-python-client)

**Describe alternatives you've considered**

[NVIDIA's DCGM Exporter](https://github.com/NVIDIA/gpu-monitoring-tools/#dcgm-exporter) but it doesn't support Jetson

rbonghi, 2021-01-22:
Hi @danmx ,

Have you read the jetson-stats documentation? https://rbonghi.github.io/jetson_stats/
you can read all jtop data from the native python API. 
There are some example available here: https://github.com/rbonghi/jetson_stats/tree/master/examples

If you have any issues, do not hesitate to write me.

closed 2021-01-22 by rbonghi

ZhenshengLee, 2024-01-07:
https://github.com/svcavallar/jetson-stats-grafana-dashboard  FYI