rbonghi/jetson_stats #129: jtop crash on Memory tab when using gnome-terminal

jankais3r, 2021-01-24:
**Describe the bug**
After upgrading to Jetpack 4.5 [L4T 32.5.0], jtop crashes every time I try to switch to the Memory tab. It only happens when running in GNOME Terminal - over SSH or in LXTerminal it doesn't crash.

**To Reproduce**
Steps to reproduce the behavior:
1. Open GNOME Terminal
2. Run jtop
3. Switch to the Memory tab
4. See error

**Traceback**
```
Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.6/dist-packages/jtop/__main__.py", line 146, in main
    curses.wrapper(JTOPGUI, jetson, [ALL, GPU, CPU, MEM, CTRL, INFO], init_page=args.page, loop=args.loop, seconds=LOOP_SECONDS)
  File "/usr/lib/python3.6/curses/__init__.py", line 94, in wrapper
    return func(stdscr, *args, **kwds)
  File "/usr/local/lib/python3.6/dist-packages/jtop/gui/jtopgui.py", line 109, in __init__
    self.run(loop, seconds)
  File "/usr/local/lib/python3.6/dist-packages/jtop/gui/jtopgui.py", line 138, in run
    self.draw()
  File "/usr/local/lib/python3.6/dist-packages/jtop/gui/lib/common.py", line 80, in wrapped
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/jtop/gui/jtopgui.py", line 153, in draw
    page.draw(self.key, self.mouse)
  File "/usr/local/lib/python3.6/dist-packages/jtop/gui/pmem.py", line 215, in draw
    self.memory_legend(line_counter + 1, height, width, first)
  File "/usr/local/lib/python3.6/dist-packages/jtop/gui/pmem.py", line 135, in memory_legend
    rectangle(self.stdscr, start_y, start_x, start_y + r_height, start_x + r_width)
  File "/usr/lib/python3.6/curses/textpad.py", line 16, in rectangle
    win.addch(lry, lrx, curses.ACS_LRCORNER)
_curses.error: addch() returned ERR
```
Add any other context about the problem here.

**Board**
 - SOC: tegra210
 - ID: 33
 - Board(s): P3542-000
 - Code Name: batuu

**Jetpack**
 - Jetpack: 4.5
 - L4T: 32.5.0

**Jetson-Stats**
 - Version: 3.0.3

```
uname -a
Linux jetson 4.9.201-tegra #1 SMP PREEMPT Fri Jan 15 14:41:02 PST 2021 aarch64 aarch64 aarch64 GNU/Linux

NVIDIA Jetson Nano (Developer Kit Version) - Jetpack 4.5 [L4T 32.5.0]

 - Up Time:        0 days 0:22:29                                                        Version: 3.0.3
 - Jetpack:        4.5 [L4T 32.5.0]                                                       Author: Raffaello Bonghi
 - Board:                                                                                 e-mail: raffaello@rnext.it
   * Type:           Nano (Developer Kit Version)
   * SOC Family:     tegra210     ID: 33
   * Module:         P3448-0003   Board: P3542-000
   * Code Name:      batuu
   * Cuda ARCH:      5.3
   * Serial Number:  1424020011274
   * Board ids:      3448
 - Libraries:                                                                             - Hostname:    jetson
   * CUDA:         10.2.89                                                                - Interfaces:
   * OpenCV:       4.1.1  compiled CUDA: NO                                                 * wlan0:     192.168.1.69
   * TensorRT:     7.1.3.0
   * VPI:          ii libnvvpi1 1.0.12 arm64 NVIDIA Vision Programming Interface library
   * VisionWorks:  1.6.0.501
   * Vulkan:       1.2.70
   * cuDNN:        8.0.0.180
```

rbonghi, 2021-01-24:
Hi @jankais3r ,
Can you try to enlarge the window? I think this bug is related to the window size.

jankais3r, 2021-01-24:
Confirmed, that was it. I initially tried it in a 640x480 VNC session, but once I increased the resolution, the problem went away. Feel free to close the issue, although having a graceful handling of this situation might be useful as 640x480 is the default resolution of VNC on L4T.

rbonghi, 2021-02-07:
I will fix this issue soon. I close this issue, is a duplicate of #75

closed 2021-02-07 by rbonghi