rbonghi/jetson_stats #130: I get error "This mode 10W does not exists" but it does

Lissanro, 2021-01-28:
**Describe the bug**
When I run jtop it refuses to run and outputs the following error:
`This mode 10W does not exists`

**To Reproduce**
I have Jetson Nano and I added 10W mode to `/etc/nvpmodel.conf`:

```
< POWER_MODEL ID=2 NAME=10W >
CPU_ONLINE CORE_0 1
CPU_ONLINE CORE_1 1
CPU_ONLINE CORE_2 1
CPU_ONLINE CORE_3 1
CPU_A57 MIN_FREQ  0
CPU_A57 MAX_FREQ 1479000
GPU_POWER_CONTROL_ENABLE GPU_PWR_CNTL_EN on
GPU MIN_FREQ 0
GPU MAX_FREQ 998400000
GPU_POWER_CONTROL_DISABLE GPU_PWR_CNTL_DIS auto
EMC MAX_FREQ 1600000000
```

Then I successfully switched to it with `sudo nvpmodel -m2` command. According to `sudo jetson_clocks --show` it works.

**Expected behavior**
I expected jtop to continue running normally. Even if does not support switching to custom nvpmodel from its own menu, I still want to observe frequencies, CPU load, power consumption, etc. I also expected that I still could switch to MAXN and 5W modes from jtop menu even if it will not show my custom 10W mode.

**Additional context**
My 10W custom mode basically limits Jetson Nano to stock frequencies. MAXN mode in my case is almost 20W mode because my Jetson Nano is overclocked, CPU@2.2GHz, GPU@1.2GHz, NVDEC/NVENC@0.9Ghz. It is stable, but when operating from battery under full load it obviously discharges it at faster rate than it would in 10W mode. 5W mode is good for conserving energy but it limits performance too much. So I needed something in between, and created custom 10W mode. I could limit frequencies without resorting to nvpmodel but it would be less convenient (I also would need to make sure all CPUs are online in case I'm switching from 5W mode). Running `sudo nvpmodel -m2` is much easier. Please note that this bug report have nothing to do with overclocking, without overclocking the issue still happens too and is caused by defining any custom nvpmodel and switching to it, as far as I can tell this always makes jtop to stop working.

**Board**
Jetson Nano B01

**Jetpack**
 - Jetpack: 4.5
 - L4T: 32.5.0

**Jetson-Stats**
 - Version: 3.0.3

Lissanro, 2021-01-30:
This error does not happen anymore, jtop now correctly displays 10W mode when it is chosen and allows me to choose from 3 nvpmodel modes including the custom one. I think this is because I rebooted. The title of this issue is still relevant though. If the reboot is necessary, I suggest changing error message from "This mode 10W does not exists" to something like "This mode 10W does not exists or you need to reboot if it was added recently" to make it clear that it is necessary to reboot after adding new mode to `nvpmodel.conf`. Otherwise it looked like jtop did not support custom modes.

rbonghi, 2021-02-07:
Hi @Lissanro ,

I never tried your setup. I will take a not for your hint! Thank you!

closed 2022-06-11 by rbonghi