rbonghi/jetson_stats #131: Display input voltage

Lissanro, 2021-01-28:
**Is your feature request related to a problem? Please describe.**
I often need to run `cat /sys/bus/i2c/drivers/ina3221x/6-0040/iio:device0/in_voltage0_input`. When experimenting with different power supplies and checking if they work well under load this is essential. In Nvidia forums I noticed that many make a mistake measuring voltage at the output of power supply without realising that Jetson Nano may not be getting it as expected, like 5.0V at power supply output and 4.5V at `in_voltage0_input`.

**Describe the solution you'd like**
Many people are not even aware they are supposed to check `in_voltage0_input` or that it even exists. So it would be very useful if jtop displayed it along with power consumption. Displaying current in mA may also be useful (knowing power and voltage, it is easy to calculate it: mW / mV = mA). Knowing current is useful because power supplies usually have current rating, the same is true for wires since voltage drop depends on current.

**Describe alternatives you've considered**
I ended up writing a script to run it in a loop every N seconds so I could monitor input voltage, but this is not as convenient as it would be if jtop displayed the input mV along with power consumption in mW.

rbonghi, 2021-02-07:
Hi @Lissanro 

This is a great idea, I will take note and implement it in the next version

kampkb, 2021-04-01:
As an addition adding support for UPS status might be a usefull feature

https://github.com/waveshare/UPS-Power-Module

rbonghi, 2023-03-17:
From jtop 4.2.0 is available the input voltage in page CTRL, area 6

[image]

closed 2023-03-17 by rbonghi