rbonghi/jetson_stats #135: Request M2 drive status and updated board info

Rotoslider, 2021-02-10:
I have a 32gb Jetson AGX with a 512gb M2 drive. Jtop shows it as a 16gb model and does not show the M2 drive.
Also the voltage supply mentioned earlier would be a great addition.
You have a nice program it should come included with the Jetson. Nice work
thank you
[image]

Rotoslider, 2021-02-10:
[image]
Really nice program

rbonghi, 2021-02-10:
Hi @Rotoslider,

About the M2 drive, I will take a look at how to implement it in jtop, the same story for #131 
About the 32GB AGX Xavier bug I will fix it soon. #87

rbonghi, 2023-01-05:
This error should now be fixed with the latest release **4.0** try to install and check again
```
sudo -H pip install -U jetson-stats
```

closed 2023-01-05 by rbonghi