rbonghi/jetson_stats #141: _curses.error: init_pair() returned ERR

zinwalin, 2021-03-03:
Jetson Nano 4GB

`jetson@nano:~$ jtop
Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 11, in <module>
    load_entry_point('jetson-stats==3.1.0', 'console_scripts', 'jtop')()
  File "/usr/local/lib/python2.7/dist-packages/jtop/__main__.py", line 148, in main
    curses.wrapper(JTOPGUI, jetson, [ALL, GPU, CPU, MEM, CTRL, INFO], init_page=args.page, loop=args.loop, seconds=LOOP_SECONDS)
  File "/usr/lib/python2.7/curses/wrapper.py", line 43, in wrapper
    return func(stdscr, *args, **kwds)
  File "/usr/local/lib/python2.7/dist-packages/jtop/gui/jtopgui.py", line 76, in __init__
    curses.init_pair(1, curses.COLOR_RED, curses.COLOR_BLACK)
_curses.error: init_pair() returned ERR
`

`jetson@nano:~$ jetson_release
 - NVIDIA Jetson Nano (Developer Kit Version)
   * Jetpack 4.5.1 [L4T 32.5.1]
   * NV Power Mode: MAXN - Type: 0
   * jetson_stats.service: active
 - Libraries:
   * CUDA: 10.2.89
   * cuDNN: 8.0.0.180
   * TensorRT: 7.1.3.0
   * Visionworks: 1.6.0.501
   * OpenCV: 4.1.1 compiled CUDA: NO
   * VPI: ii libnvvpi1 1.0.15 arm64 NVIDIA Vision Programming Interface library
   * Vulkan: 1.2.70`

rbonghi, 2021-03-04:
Thank you @zinwalin I will check soon

Mikystars, 2022-03-11:
I get the exact same error. Any updates?

rbonghi, 2023-01-04:
I think, like issue #123, you need to run before jtop `export TERM='xterm-256color'`

closed 2023-01-04 by rbonghi